chipsec/chipsec #328: S3BootScript Error: failed at the decoding part

IoTexplo, 2017-12-24:
I install the chipsec on my ubuntu, and I was tring to run the boot script module using the following command: "chipsec_main -m common.uefi.s3bootscript" and here is the output of it:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.0
[CHIPSEC] Arguments: -m common.uefi.s3bootscript
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.4.0-104-generic #127-Ubuntu SMP Mon Dec 11 12:16:42 UTC 2017 x86_64
[CHIPSEC] Platform: Desktop 4th Generation Core Processor (Haswell CPU / Lynx Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0C00
 
[+] loaded chipsec.modules.common.uefi.s3bootscript
[*] running loaded modules ..

[*] running module: chipsec.modules.common.uefi.s3bootscript
[x][ =======================================================================
[x][ Module: S3 Resume Boot-Script Protections
[x][ =======================================================================
[*] SMRAM: Base = 0x00000000DC000000, Limit = 0x00000000DCFFFFFF, Size = 0x01000000
[!] Found 1 S3 boot-script(s) in EFI variables
[*] Checking S3 boot-script at 0x00000000DA7E0018
[-] S3 boot-script is not in SMRAM
[*] Reading S3 boot-script from memory..
[*] Decoding S3 boot-script opcodes..
ERROR: The module was not able to recognize the S3 resume boot script on this platform.

Why it's fail at the decoded stage? Anyone can help me with this

ZandroidX, 2018-02-11:
Don't bother this infects your firmware with malware.

johnloucaides, 2018-02-12:
If you have concerns about chipsec affecting the system, please update in #344.

c7zero, 2018-02-13:
@IoTexplo the format of the S3 boot script is often firmware vendor specific. The error is probably because chipsec couldn't parse some of the opcodes in the boot script. Running with option "-v" can tell what was the parsing error. Which system vendor/model had this error?

However, even looking at your output, it looks as the boot script is not protected, so the module would likely have FAILED.

ErikBjorge, 2018-05-21:
Closing due to lack of additional information on the specific format for this resume script.  Please reopen the issue if needed.

closed 2018-05-21 by ErikBjorge