chipsec/chipsec #329: What's up with the GBe region and "Certain SPI flash regions are writeable by software"?

mjudell, 2018-01-08:
--[ 1. The Question

I ran a vanilla "$chipsec_util", which warned me that some SPI flash regions are writeable by software. This was the GBe region. (Full chipsec.modules.common.spi_access log at bottom)

I then dumped an image via "$chipsec_util spi dump firmware.bin", and decoded it via "$chipsec_util decode firmware.bin uefi".  (For what it's worth, the .bin largely agrees with the image I pulled manually using a Dediprog, and the decoded summary largely agrees with UEFITool). 

Inside the resulting firmware.bin.dir I found evidence of three regions "Flash Descriptor", "BIOS", and "Intel ME". By evidence, I mean paired bin/log files and directories.

However, inside "...Flash Descriptor.bin.log" I saw the results below.  Does this mean GBe and other regions exist only in the flash descriptor, but not elsewhere on the EEPROM's? If so, should the warning message include some indication that "yeah, GBe is read/write but its base is higher than its limit, so it doesn't really exist."

Why is there no GBe region here?  Does that mean GBe firmware is stored in another region, or in software? fwiw, I have a Realtek (not Intel) GBe controller.  

--[ 2. Flash Descriptor Region Summary  
`Flash Regions
--------------------------------------------------------
 Region                | FLREGx    | Base     | Limit   
--------------------------------------------------------
0 Flash Descriptor     | 00000000  | 00000000 | 00000000 
1 BIOS                 | 0BFF0400  | 00400000 | 00BFF000 
2 Intel ME             | 03FF0001  | 00001000 | 003FF000 
3 GBe                  | 00007FFF  | 07FFF000 | 00000000 (not used)
4 Platform Data        | 00007FFF  | 07FFF000 | 00000000 (not used)
8 Embedded Controller  | 00007FFF  | 07FFF000 | 00000000 (not used)

+ 0x0080 Master Section:
========================================================
+ 0x0080 FLMSTR0   : 0x00A00B00
+ 0x0084 FLMSTR1   : 0x00C00D00

Master Read/Write Access to Flash Regions
--------------------------------------------------------
 Region                | CPU   | ME    
--------------------------------------------------------
0 Flash Descriptor     | R     | R     
1 BIOS                 | RW    |       
2 Intel ME             |       | RW    
3 GBe                  | RW    | RW    
4 Platform Data        |       |       
5 Flash Region 5       |       |       
6 Flash Region 6       |       |       
7 Flash Region 7       |       |       
8 Embedded Controller  |       |     
`

--[ 3. Full chipsec.modules.common.spi_access Log

`[*] running module: chipsec.modules.common.spi_access
[x][ =======================================================================
[x][ Module: SPI Flash Region Access Control
[x][ =======================================================================
SPI Flash Region Access Permissions
------------------------------------------------------------
[*] FRAP = 0x00004ACB << SPI Flash Regions Access Permissions Register (SPIBAR + 0x50)
    [00] BRRA             = CB << BIOS Region Read Access 
    [08] BRWA             = 4A << BIOS Region Write Access 
    [16] BMRAG            = 0 << BIOS Master Read Access Grant 
    [24] BMWAG            = 0 << BIOS Master Write Access Grant 

BIOS Region Write Access Grant (00):
  FREG1_BIOS  : 0
  FREG2_ME    : 0
  FREG3_GBE   : 0
BIOS Region Read Access Grant (00):
  FREG1_BIOS  : 0
  FREG2_ME    : 0
  FREG3_GBE   : 0
BIOS Region Write Access (4A):
  FREG0_FLASHD: 0
  FREG1_BIOS  : 1
  FREG2_ME    : 0
  FREG3_GBE   : 1
  FREG4_PD    : 0
  FREG5       : 0
  FREG6       : 1
BIOS Region Read Access (CB):
  FREG0_FLASHD: 1
  FREG1_BIOS  : 1
  FREG2_ME    : 0
  FREG3_GBE   : 1
  FREG4_PD    : 0
  FREG5       : 0
  FREG6       : 1
[!] WARNING: Software has write access to GBe region in SPI flash
[!] WARNING: Certain SPI flash regions are writeable by software`

ErikBjorge, 2018-01-08:
I think we need to something better in spi_access to cover this case.  In the image that you have the GbE region is disabled.  If it was enabled it would reside in the SPI flash device and would be writable.  I think that the spi_access module needs to be updated to check if the region is enabled as well as if write access is enabled.

I will mark this as a bug so we can improve the test.

ZandroidX, 2018-02-11:
Do not run mmio rogue it overwrites your sound and wifi firmware so you'll have low grade audio and internet speed. Chipsec is malware.

mjudell, 2018-02-12:
ZandroidX, that is a remarkable claim.  It might be helpful for you to point out the allegedly malicious payload in the repo, if you're not trolling.

ZandroidX, 2018-02-12:
Ok what do you need from me I'd really appreciate the help!

johnloucaides, 2018-02-12:
I've just opened issue #344 for this. Can you please update in that issue what exactly was run and what symptoms you see?

Also, if you have suspicions about specific files in the repository, please feel free to add that in #344 as well. 

Note that chipsec is a test tool and not intended for use on production systems. We have marked it such in multiple places. Particularly the fuzzer tools may actually cause undesirable behavior. This is sometimes necessary in order to discover/understand new issues. These tools should only be run at your own risk.

ZandroidX, 2018-02-12:
Yes it was running the fuzzer when I began noticing these symptoms, I can't even remember the addresses or offset used was like f:0;00505 or 0a0a

ZandroidX, 2018-02-12:
ran rogue smrr and the fuzzer

ErikBjorge, 2018-02-13:
@ZandroidX can you please add your system configuration to the issue that @johnloucaides opened?  We can then help figure out what is going on.  Can you also include the model/processor information and system configuration data such as OS version and anything else (like log files) that may help.

Issue Link:
https://github.com/chipsec/chipsec/issues/344

c7zero, 2018-02-13:
commented in #344

ZandroidX, 2018-02-13:
Lenovo T440 Type 2B06
Intel I7 4600U
Running windows 10
BIOS Version/Date	LENOVO GJET94WW (2.44 ), 9/14/2017
SMBIOS Version	2.7

Reporting same Issue as original post.
at random times when I click it goes to lower left hand corner of screen pos(x =0, y=0)
cannot furnish a log because I'm not running Ubuntu now.
I am running Hyper-v
WiFi Is downloading really slow and dropping connection at random times.
Windows update KB4054517 noit installing says awaiting restart every boot.

akashdeep-aulakh, 2018-03-30:
ErikBjorge, this should also be platform specific. For the PCHs where the GBe region is read/write there is no reason for the spi_access module to complain - for example Lewisburg (agree?)

ErikBjorge, 2018-03-30:
The expected setting is specific to the platform generation being tested.  We just need to define a clean method of specifying the expected programming state for the specific platform.  In addition, if the region is disabled we should not bother checking the state and list it as disabled (information only).

Sae86, 2022-06-14:
Closing since this is working as designed.

closed 2022-06-14 by Sae86