chipsec/chipsec #334: chipsec_util trigger out of memory killer

pietrushnic, 2018-01-22:
I'm trying CHIPSEC from `master` installed in Python virtualenv using Debian stable on MinnowBoard Turbot. `chipsec_util acpi list` cause:

```
[  597.439161] chipsec: loading out-of-tree module taints kernel.
[  597.447858] Chipsec module loaded
[  597.451710] ** This module exposes hardware & memory access, **
[  597.458349] ** which can effect the secure operation of      **
[  597.464975] ** production systems!! Use for research only!   **
[  597.682916] ioremap error for 0x78c3d000-0x78c3e000, requested 0x2, got 0x0
[  597.690747] ioremap error for 0x78c3e000-0x78c3f000, requested 0x2, got 0x0
[  597.699173] ioremap error for 0x78c3f000-0x78c40000, requested 0x2, got 0x0
[  597.707410] ioremap error for 0x78c40000-0x78c41000, requested 0x2, got 0x0
[  597.715660] ioremap error for 0x78c42000-0x78c43000, requested 0x2, got 0x0
[  597.730432] ioremap error for 0x78cae000-0x78caf000, requested 0x2, got 0x0
[  597.741404] ioremap error for 0x78ce9000-0x78cea000, requested 0x2, got 0x0
[  597.749888] ioremap error for 0x78cea000-0x78ceb000, requested 0x2, got 0x0
[  597.757718] ioremap error for 0x78ceb000-0x78cec000, requested 0x2, got 0x0
[  597.765957] ioremap error for 0x78cec000-0x78ced000, requested 0x2, got 0x0
[  597.773803] ioremap error for 0x78ced000-0x78cee000, requested 0x2, got 0x0
[  597.781684] ioremap error for 0x78cee000-0x78cef000, requested 0x2, got 0x0
[  597.789538] ioremap error for 0x78cef000-0x78cf0000, requested 0x2, got 0x0
[  597.797394] ioremap error for 0x78cf0000-0x78cf1000, requested 0x2, got 0x0
[  597.805241] ioremap error for 0x78cf1000-0x78cf2000, requested 0x2, got 0x0
[  597.813091] ioremap error for 0x78cf2000-0x78cf3000, requested 0x2, got 0x0
[  597.820943] ioremap error for 0x78cf3000-0x78cf4000, requested 0x2, got 0x0
[  597.828806] ioremap error for 0x78cf4000-0x78cf5000, requested 0x2, got 0x0
[  597.836663] ioremap error for 0x78cf5000-0x78cf6000, requested 0x2, got 0x0
[  597.844515] ioremap error for 0x78cf6000-0x78cf7000, requested 0x2, got 0x0
[  597.852370] ioremap error for 0x78cf7000-0x78cf8000, requested 0x2, got 0x0
[  597.860218] ioremap error for 0x78cf8000-0x78cf9000, requested 0x2, got 0x0
[  597.868071] ioremap error for 0x78cf9000-0x78cfa000, requested 0x2, got 0x0
[  597.875926] ioremap error for 0x78cfa000-0x78cfb000, requested 0x2, got 0x0
[  597.883777] ioremap error for 0x78cfb000-0x78cfc000, requested 0x2, got 0x0
[  598.015532] ioremap error for 0x78390000-0x78391000, requested 0x2, got 0x0
[  610.649136] chipsec_util invoked oom-killer: gfp_mask=0x26040c0(GFP_KERNEL|__GFP_COMP|__GFP_NOTRACK), nodemask=0, order=0, oom_score_adj=0
[  610.663108] chipsec_util cpuset=/ mems_allowed=0
[  610.668319] CPU: 0 PID: 7909 Comm: chipsec_util Tainted: G           O    4.9.0-4-amd64 #1 Debian 4.9.65-3+deb9u1
[  610.679791] Hardware name: ADI Minnowboard Turbot D0 PLATFORM/MinnowBoard Turbot, BIOS MNW2MAX1.X64.0097.D01.1709211100 09/21/2017
[  610.692916]  0000000000000000 ffffffffa7d29134 ffffb7e701257c80 ffff91bcf6bf2f00
[  610.701196]  ffffffffa7bfe340 0000000000000000 0000000000000000 0101000c00000001
[  610.709481]  502fde7296ec4753 00000000ffffffff 0000000000000000 0000000000000000
[  610.717759] Call Trace:
[  610.720498]  [<ffffffffa7d29134>] ? dump_stack+0x5c/0x78
[  610.726437]  [<ffffffffa7bfe340>] ? dump_header+0x78/0x1fd
[  610.732570]  [<ffffffffa7b804aa>] ? oom_kill_process+0x21a/0x3e0
[  610.739285]  [<ffffffffa7b8012d>] ? oom_badness+0xed/0x170
[  610.745417]  [<ffffffffa7b80941>] ? out_of_memory+0x111/0x470
[  610.751841]  [<ffffffffa7b85bb7>] ? __alloc_pages_slowpath+0xbb7/0xbf0
[  610.759140]  [<ffffffffa7b85dee>] ? __alloc_pages_nodemask+0x1fe/0x260
[  610.766438]  [<ffffffffa7bde3cd>] ? cache_grow_begin+0x9d/0x550
[  610.773055]  [<ffffffffa7bde3cd>] ? cache_grow_begin+0x9d/0x550
[  610.779673]  [<ffffffffa7bdeb31>] ? fallback_alloc+0x161/0x200
[  610.786200]  [<ffffffffc083d51b>] ? read_mem+0xdb/0x230 [chipsec]
[  610.793012]  [<ffffffffa7bdf9c1>] ? __kmalloc+0x1e1/0x560
[  610.799048]  [<ffffffffc083d51b>] ? read_mem+0xdb/0x230 [chipsec]
[  610.805861]  [<ffffffffa7c01f91>] ? vfs_read+0x91/0x130
[  610.811700]  [<ffffffffa7c03462>] ? SyS_read+0x52/0xc0
[  610.817445]  [<ffffffffa800763b>] ? system_call_fast_compare_end+0xc/0x9b
[  610.825054] Mem-Info:
[  610.827606] active_anon:2128 inactive_anon:2416 isolated_anon:0
[  610.827606]  active_file:1672 inactive_file:1717 isolated_file:0
[  610.827606]  unevictable:0 dirty:1127 writeback:2761 unstable:0
[  610.827606]  slab_reclaimable:2999 slab_unreclaimable:454921
[  610.827606]  mapped:1756 shmem:203 pagetables:553 bounce:0
[  610.827606]  free:13116 free_pcp:173 free_cma:0
[  610.865039] Node 0 active_anon:8512kB inactive_anon:9664kB active_file:6688kB inactive_file:6868kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:7024kB dirty:4508kB writeback:11044kB shmem:812s
[  610.897893] Node 0 DMA free:7792kB min:372kB low:464kB high:556kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15984kB managed:15892kB mlocked:B
[  610.931436] lowmem_reserve[]: 0 1859 1859 1859 1859
[  610.936918] Node 0 DMA32 free:44672kB min:44680kB low:55848kB high:67016kB active_anon:8512kB inactive_anon:9664kB active_file:6688kB inactive_file:6868kB unevictable:0kB writepending:15552kB present:1970944B
[  610.974938] lowmem_reserve[]: 0 0 0 0 0
[  610.979257] Node 0 DMA: 0*4kB 0*8kB 1*16kB (U) 1*32kB (U) 1*64kB (U) 0*128kB 0*256kB 1*512kB (U) 1*1024kB (U) 1*2048kB (M) 1*4096kB (M) = 7792kB
[  610.993863] Node 0 DMA32: 3092*4kB (ME) 1530*8kB (UME) 626*16kB (UME) 242*32kB (UME) 34*64kB (UM) 1*128kB (M) 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 44672kB
[  611.010217] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
[  611.019659] 6027 total pagecache pages
[  611.023853] 2435 pages in swap cache
[  611.027854] Swap cache stats: add 8514, delete 6079, find 37/63
[  611.034478] Free swap  = 1951728kB
[  611.038282] Total swap = 1985532kB
[  611.042078] 496732 pages RAM
[  611.045303] 0 pages HighMem/MovableOnly
[  611.049593] 11115 pages reserved
[  611.053204] 0 pages hwpoisoned
[  611.056611] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[  611.066150] [  187]     0   187    12741      135      25       3      127             0 systemd-journal
[  611.076762] [  207]     0   207    11549       12      23       3      331         -1000 systemd-udevd
[  611.087171] [  276]   100   276    31821        0      31       4      136             0 systemd-timesyn
[  611.097782] [  300]     0   300     7409        1      19       3       63             0 cron
[  611.107322] [  302]     0   302    62529        0      28       3      287             0 rsyslogd
[  611.117254] [  304]   105   304    11280        0      26       3      115          -900 dbus-daemon
[  611.127476] [  309]     0   309    11605       13      26       3      126             0 systemd-logind
[  611.137989] [  320]     0   320    17360      129      38       3        2             0 login
[  611.147625] [  322]     0   322     3634        0      11       3       39             0 agetty
[  611.157362] [  330]     0   330    14106       68      31       3      131             0 systemd
[  611.167193] [  331]     0   331    20592        0      43       3      371             0 (sd-pam)
[  611.177123] [  333]     0   333     5230      114      15       3      289             0 bash
[  611.186663] [  351]     0   351     5118        5      12       3      256             0 dhclient
[  611.196593] [  545]     0   545    17486        0      38       3      192         -1000 sshd
[  611.206135] [  546]     0   546    23801      184      51       3       59             0 sshd
[  611.215676] [  552]     0   552     5238        1      14       3      409             0 bash
[  611.225216] [ 1654]     0  1654     4833      172      16       3       82             0 tmux: client
[  611.235534] [ 1656]     0  1656     7123      195      21       3      244             0 tmux: server
[  611.245853] [ 1657]     0  1657     5320        1      14       3      493             0 bash
[  611.255393] [ 7909]     0  7909    17414     2579      39       3     3407             0 chipsec_util
[  611.265711] Out of memory: Kill process 7909 (chipsec_util) score 5 or sacrifice child
[  611.274575] Killed process 7909 (chipsec_util) total-vm:69656kB, anon-rss:5040kB, file-rss:5276kB, shmem-rss:0kB
[  611.289466] oom_reaper: reaped process 7909 (chipsec_util), now anon-rss:72kB, file-rss:0kB, shmem-rss:0kB
```

Any idea what is wrong ?

ErikBjorge, 2018-01-23:
I am not sure what is happening.  I was able to reproduce the issue on a Turbot that I have.  I will see if this happens on other systems I have.

ErikBjorge, 2018-01-31:
I have tried a few other systems and this only seems to be happening on the Turbot system.  I will need to do some more investigation of the issue.

ErikBjorge, 2018-12-21:
Need to retest this with the latest version of CHIPSEC to see if this is still happening.

npmitche, 2023-05-11:
I believe this is fixed, but am not able to locate a minnowboard to verify. If this is still an issue, please re-open an issue.

closed 2023-05-11 by npmitche