chipsec/chipsec #335: Question about  (D31:F5)

xcompon3nt, 2018-01-23:
I guess there is something I'm missing, could anyone explain why is BC on (b:d.f 00:31.5 + 0xDC) and not on (b:d.f 00:31.0 + 0xDC) in accordance with every datasheet I've stumbled on? What I'm I missing, thanks in advance!

skochinsky, 2018-01-23:
What is BC and where did you get 00:31.5 from? Also, how is it related to chipsec?

xcompon3nt, 2018-01-23:
[*] running module: chipsec.modules.common.bios_wp
[x][ =======================================================================
[x][ Module: BIOS Region Write Protection
[x][ =======================================================================
[*] BC = 0x00000A88 << BIOS Control (b:d.f 00:31.5 + 0xDC)
    [00] BIOSWE           = 0 << BIOS Write Enable 
    [01] BLE              = 0 << BIOS Lock Enable 
    [02] SRC              = 2 << SPI Read Configuration 
    [04] TSS              = 0 << Top Swap Status 
    [05] SMM_BWP          = 0 << SMM BIOS Write Protection 
    [06] BBS              = 0 << Boot BIOS Strap 
    [07] BILD             = 1 << BIOS Interface Lock Down 
[-] BIOS region write protection is disabled!

I'm confused. It's z270, although I don't think that matter, shouldn't it be F0 on all PCH?

pietrushnic, 2018-01-23:
@xcompon3nt according to [datasheet](https://www.intel.com/content/www/us/en/chipsets/200-series-chipset-pch-datasheet-vol-1.html) this is correct D31:F5. In [datasheet vol.2](https://www.intel.com/content/www/us/en/chipsets/200-series-chipset-pch-datasheet-vol-2.html) you can confirm that `0xDC` is correct offset for `BIOS Control (BIOS_SPI_BC)` register.

xcompon3nt, 2018-01-23:
I still don't understand how the BIOS Control register (0xDC), which is part of the LPC (or eSPI) =  D31:F0 and not part of the SPI (D31:F5),  examined in chipsec as D31:F5:DC.
Excuse my ignorance, I'm just trying to understand, it seems to be some kind of confusion on part of me and my colleagues, mainly because of this:
[image]

pietrushnic, 2018-01-23:
@xcompon3nt my understanding is that availability of BIOS Control register on LPC is due to backward compatibility feature and possibility of using SPI device connected on LPC bus (aka FWH) or eSPI as boot firmware location. From what I see BIOS Control is defined for LPC, eSPI and SPI. Please note that those registers are different. First is just 8bit and expose configuration about BIOS memory range decoding and what should be used (LPC or eSPI). Second (eSPI BIOS Control register) is 32 bit but have just 2 bits defined related to writing to eSPI. Last expose full information about things CHIPSEC want to check and there is silent assumption that by default SPI is boot firmware location.

I see that some features are in both registers e.g. Boot BIOS Strap (BBS). I'm not sure why this confusion was introduced (maybe copy&paste issue). I'm also not sure what will happen if you configure those registers in different way e.g. BBS in LPC to 0 and BBS in SPI to 1. This kind of deep analysis and clarification should be addressed to Intel Embedded Community.

xcompon3nt, 2018-01-23:
Yes, I do understand now. But I don't understand why is the BIOS_SPI_BC used and not BC. I guess I'll have to try both in my module to see which one will 'react' as described.

xcompon3nt, 2018-01-23:
My apologies, I've built a DXE driver and tested both, BIOS_SPI_BC is correct. BC does absolutely nothing except for the SMI generation on WPD set while LE. I've also noticed that the description for the InSMM.STS is just a bit clearer for BIOS_SPI_BC (possibly that's the updated version).
Sorry for wasting your time, it's all clear now, thank you!

closed 2018-05-21 by ErikBjorge