chipsec/chipsec #339: `ImportError: cannot import name efi_compressor`

paulmenzel, 2018-01-29:
With Ubuntu 17.10 after doing `sudo python setup.py install`, some modules are still missing.

```
$ git log --oneline -1
527f3d1 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #330 from c7zero/spectre
$ sudo python chipsec_main.py --version
Traceback (most recent call last):
  File "chipsec_main.py", line 56, in <module>
    import chipsec.module
  File "/home/helmuth/src/chipsec/chipsec/module.py", line 27, in <module>
    from chipsec.module_common import ModuleResult
  File "/home/helmuth/src/chipsec/chipsec/module_common.py", line 43, in <module>
    import chipsec.chipset
  File "/home/helmuth/src/chipsec/chipsec/chipset.py", line 41, in <module>
    from chipsec.helper.oshelper import OsHelper, OsHelperError
  File "/home/helmuth/src/chipsec/chipsec/helper/oshelper.py", line 93, in <module>
    import chipsec.helper.helpers
  File "/home/helmuth/src/chipsec/chipsec/helper/helpers.py", line 26, in <module>
    from chipsec.helper.linux import *
  File "/home/helmuth/src/chipsec/chipsec/helper/linux/helper.py", line 56, in <module>
    from chipsec_tools import efi_compressor
ImportError: cannot import name efi_compressor
```

abazhaniuk, 2018-01-31:
Could you show log from sudo python setup.py install , please?

eclypsium, 2018-02-06:
```python setup.py build_ext -i```
should help

closed 2018-02-06 by abazhaniuk