chipsec/chipsec #340: Device 8086:1614

abazhaniuk, 2018-01-30:
Don't have Device 8086:1614 in chipset.py
https://pci-ids.ucw.cz/read/PC/8086/1614

abazhaniuk, 2018-01-31:
This is interesting issue because i didn't find db with all vid/did, what i found is in https://pci-ids.ucw.cz : 

```
$ grep "broadwell" pci.ids.txt  -i| grep 
Bridge
	1600  Broadwell-U Host Bridge -OPI
	1604  Broadwell-U Host Bridge -OPI
	1608  Broadwell-U Host Bridge -OPI
	1610  Broadwell-U Host Bridge - DMI
	1614  Broadwell-U Host Bridge - DMI
	1618  Broadwell-U Host Bridge - DMI
$ grep "haswell" pci.ids.txt  -i| grep Bridge
	2f0d  Haswell Xeon Non-Transparent Bridge (Back-to-back)
	2f0e  Haswell Xeon Non-Transparent Bridge (Primary Side)
	2f0f  Haswell Xeon Non-Transparent Bridge (Secondary Side)
$ 
```

Which showing that chipsec doesn't support a lot of chipsets.... 

Do you know where we can get all vid/did for specific gen, 
for example vid/did for all hsw of all laptop, desktop

skochinsky, 2018-01-31:
AFAIK there's no such table besides pci-ids but Intel's processor Spec updates often have a list of device IDs for the PCI device D0F0 (Host Bridge) and sometime others for a specific CPU line. E.g.:

 - [4th gen desktop](https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/4th-gen-core-family-desktop-specification-update.pdf)
- [5th gen](https://www3.intel.com/content/dam/www/public/us/en/documents/specification-updates/5th-gen-core-family-spec-update.pdf)
 - [6th gen](https://www3.intel.com/content/dam/www/public/us/en/documents/specification-updates/desktop-6th-gen-core-family-spec-update.pdf)
- [7th/8th gen](https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/7th-gen-core-family-spec-update.pdf)

Processor datasheet vol. 2 ([e.g.](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/5th-gen-core-family-datasheet-vol-2.pdf)) also usually has some device IDs.

More: https://duckduckgo.com/?q=%22Processor+Identification+by+Register+Contents%22+intel and https://duckduckgo.com/?q="PCI+Devices+and+Functions"+site%3Aintel.com

abazhaniuk, 2018-02-18:
fixed: https://github.com/chipsec/chipsec/pull/346

closed 2018-02-18 by abazhaniuk