chipsec/chipsec #344: Investigate Reported Issues in rogue_mmio_bar.py

johnloucaides, 2018-02-12:
Placeholder for reported issues after running this test

c7zero, 2018-02-13:
Ok, this sounds more like someone is testing a generator of comments for Github issues but I'll pretend that this is a real person having a real issue after running [tools.smm.rogue_mmio_bar](https://github.com/chipsec/chipsec/blob/master/chipsec/modules/tools/smm/rogue_mmio_bar.py)  module.

tools.smm.rogue_mmio_bar module checks for issues in SMI handlers described in [this RECon presentation](http://www.c7zero.info/stuff/REConBrussels2017_BARing_the_system.pdf)  (www.intelsecurity.com link in the module source code no longer works). This kind of issues are due to the fact that some SMI handlers may write to MMIO registers in ranges located at addresses defined by BAR PCIe configuration registers that cannot be locked down. By changing the address in BAR register and invoking an SMI attacker can trick SMI handler to write to attacker controlled address potentially corrupting SMRAM.

What the module does is it modifies BAR registers of specified PCIe device or all devices chipsec finds to point to some allocated memory region and triggers whole bunch of SMIs. Then the module checks if contents at that memory range changed indicating that some of the SMI handlers did write to that range. 

Can it crash the system/OS? - Sure. It relocates MMIO ranges underneath the device. That's why it's in tools.*
Can it brick the system? - Unlikely, it doesn't access any firmware or flash. It only relocates MMIO ranges in memory. But if the firmware on a system is dumb enough to write to the flash whatever data supplied to one of its SMI handlers - then this would write garbage to the flash. Any module which fuzzes SMIs would cause this. It may be worth adding a warning waiting for command line input to start the test, just like in tools.uefi.uefivar_fuzz module so that nobody even accidentally corrupts their flash.

You should understand that chipsec talks to devices over various hardware interfaces so it can brick systems if you don't know what you are doing (eg. running commands like "chipsec_util spi erase"). Maybe we should also consider adding a warning expecting command-line input when running chipsec_util by default.

It's hard to take seriously the claim about malware... All commits (exactly 1) are in the history.

ZandroidX, 2018-02-13:
That's probably what happened on mine. The MMIO registers were all relocated and now I'm experiencing some kind of peg in system performance. Any way to reset the MMIO Registers to default using the PCH?

https://www.intel.com/content/www/us/en/chipsets/9-series-chipset-pch-datasheet.html

skochinsky, 2018-02-14:
@ZandroidX did reboot/power-down not restore them? I don't think chipsec is able to do any kind of permanent changes in the firmware configuration so a reset should fix it (unless the original firmware is buggy in some weird way).

ZandroidX, 2018-02-15:
The original firmware is infected, it reflashes bios (acpi etc.) Before an os is even loaded, now, it's a matter of finding out how it reflashing the bios via spi or chipset intel series 9 pch badbios restore is infected

skochinsky, 2018-02-15:
Ok, so it's not a chipsec issue then?

ZandroidX, 2018-02-16:
It was a chipsec issue but c7zero answered my question about the mmio ranges being readdressed. You can mark this thread as resolved.

closed 2018-04-24 by johnloucaides