chipsec/chipsec #351: spi_desc chipsec  issues

abazhaniuk, 2018-02-18:
Looks like on some systems (guess SKL+) has issue in spi_desc functionality in chipsec. 
result from chipsec_util.py spidesc and chipsec_main.py -m common.spi_desc - not consistent. result is different from chipsec_util.py spidesc and chipsec_main.py -m common.spi_desc

ErikBjorge, 2018-03-02:
Can you provide the log files for each case so I can do some debugging on the issue?

abazhaniuk, 2018-03-27:
Issue reproduced. 
We have 2 format for SPI Descriptor: common, skl+ 
When you run common (in any system older then skl) with skl image, you will see warning that probably you have image with SKL descriptor. But then you run on skl+ system with old image you will not see any warning but descriptor will be decoded incorrectly.

closed 2026-10-12 by npmitche