chipsec/chipsec #358: BIOS, ME/BKC and Chipsec versions should be tied together.

akashdeep-aulakh, 2018-02-23:
BIOS, ME/BKC and Chipsec versions should be tied together. This will help enforce, as requirement, that an expected version of Chipsec shows expected results against particular BIOS, ME/BKC. This will help better enforce Chipsec as a quality assessment requirement.

johnloucaides, 2018-02-23:
Thanks for adding this here! In order to implement, I think we need informational modules for BIOS version (parse SMBIOS table to get this) and ME version (need HECI HAL to get this). We're already working on some of this.

akashdeep-aulakh, 2018-03-27:
Wondering how this is going?

ErikBjorge, 2018-03-29:
We are going through some internal review of the ME support code.  Once we think the code is clean enough we will submit it for open review.  We have not yet started the support to extract the SMBIOS information yet.

npmitche, 2024-01-31:
I don't see a need to correlate the three together in this current landscape. If anyone thinks this is still needed, please open a new issue. With that said, I would like to add an info module that prints the bios version string from SMBIOS, but that should be pretty straight forward as we already have that functionality in the `chipsec_util.py smbios get decoded` tool

closed 2024-01-31 by npmitche