chipsec/chipsec #363: Kernel module does not load in Fedora running kernel v4.15.4-300 and above

ettermyra, 2018-03-01:
When loading the CHIPSEC kernel module in Fedora kernel 4.15.4-300 and above the error "Unknown symbol in module" is produced. This applies to at least version 1.3.6RC1 and 1.3.5 of CHIPSEC.

When loading in Fedora kernel 4.14.16 it all works just fine! Unfortunately I do not know what has changed from one kernel version to another...Some patches related to Meltdown and Spectre?...just specualting since I have not read the release notes for the kernel.

johnloucaides, 2018-03-01:
I'll try this when I get to my Fedora system, but can you copy/paste which symbol is "unknown"?

ettermyra, 2018-03-01:
Oh! Sorry about that! =P Here goes (dmesg):
Unknown symbol valid_mmap_phys_addr_range (err 0)

petersandor, 2018-03-25:
I'm experiencing the same issue on Arch 4.15.12-1

ErikBjorge, 2018-03-27:
Thanks for the information @petersandor.  I am going to have @BrentHoltsclaw take a look at it.

BrentHoltsclaw, 2018-04-03:
I created a pull request Fix for issue #363.  If you have time, could you check to see if it fixes your issue.

ettermyra, 2018-04-18:
Tried the suggested fix and initial results indicate that it works fine. Will do some more testing...

johnloucaides, 2018-04-24:
Are we ok to get this merged? Kinda annoying to have Fedora broken. ;-)

ErikBjorge, 2018-04-25:
It was merged on the 17th.

johnloucaides, 2018-04-25:
My bad. It wasn't this error that I noticed. It was that Fedora 27 now needs *another* dependency (elfutils-libelf-devel). Sorry.

I've updated the linux install instructions here: https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-Linux

ettermyra, 2018-04-28:
Sorry for the delay!
All seem to be working just fine! Closing this issue.
Thanks!

closed 2018-04-28 by ettermyra