chipsec/chipsec #365: Specify -p --platform causes extra failed modules

weitang730, 2018-03-02:
I am using chipsec 1.3.6 rc1 plus p0005 on a Purley platform with Skylake CPUs.
Run "sudo chipsec_main -p SKL" causes 4 more failures.
[CHIPSEC] Modules failed        4:
[-] FAILED: chipsec.modules.smm_dma <----------------------
[-] FAILED: chipsec.modules.mclock <-----------------------
[-] FAILED: chipsec.modules.remap <-----------------------
[-] FAILED: chipsec.modules.memconfig <----------------------- 
However "sudo chipsec_main" without specific platform works fine. 
Which results should I take?

ErikBjorge, 2018-03-02:
You should only use the -p flag if you need to override the automatic detection process used when just running chipsec_main.  The results from just running chipsec_main should be used unless the platform can't be detected.

Also, most server (E5/E7) parts will end in an X.  So in this case it should be loading the SKX configuration files.

weitang730, 2018-03-05:
Thanks a lot!

closed 2018-03-05 by weitang730