chipsec/chipsec #367: BIOS parsing issue

abazhaniuk, 2018-03-06:
I think we have issue in decoding/parsing. Here is example:
[FW.zip](https://github.com/chipsec/chipsec/files/1786274/FW.zip)

FW: ```FW.bin.dir/1_700000-FFFFFF_BIOS.bin.dir/FV/03_5473C07A-3DCB-4DCA-BD6F-1E9689E7349A.dir/00_9E21FD93-9C72-4C15-8C4B-E77F1DB2D792.FV_FVIMAGE.dir/00_S_GUID_DEFINED.dir/01_S_FV_IMAGE```
not decoding.

ErikBjorge, 2018-03-06:
Thanks for providing the FW file for us to look at.  I think I will have @BrentHoltsclaw look at the issue and get back to you.

BrentHoltsclaw, 2018-05-22:
The FW provided fails parsing due to the way chipsec handles parsing sec.Type EFI_SECTION_FIRMWARE_VOLUME_IMAGE. 
NextFwVolume (within build_efi_tree) treats the remaining data as a firmware header.  Please see the attached log of running the FW with VolInfo from edk BaseTools. 

[Volinfo.txt](https://github.com/chipsec/chipsec/files/2027741/Volinfo.txt)

abazhaniuk, 2018-06-06:
Any suggested fix?

ErikBjorge, 2018-06-06:
At this time it looks like we need to update/rewrite the FV parsing code.  We do not have an ETA for this change yet.  We should also make sure the FV parsing code matches the process implemented in the EDK2 source so CHIPSEC behaves the same as EDK2.

BrentHoltsclaw, 2018-06-08:
Working on a fix, hopefully will have something out next week for you to test

ErikBjorge, 2018-12-21:
Closing as this should be fixed with latest release and updated decompression code.  Please reopen if additional support is required.

closed 2018-12-21 by ErikBjorge