chipsec/chipsec #374: Exception when running chipsec_util uefi var-list

martingalvan, 2018-04-16:
When running this command on Ubuntu 17.10 I get the following error:

```
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Executing command 'uefi' with args ['var-list']

[CHIPSEC] Enumerating all EFI variables via OS specific EFI Variable API..
[CHIPSEC] Decoding EFI Variables..
Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.3.6rc1', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec_util.py", line 258, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec_util.py", line 225, in main
    comm.run()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 151, in run
    decode_EFI_variables( efi_vars, nvram_pth )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 231, in decode_EFI_variables
    parse_efivar_file( var_fname, data )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi_common.py", line 835, in parse_efivar_file
    parse_db( var, var_path )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi_common.py", line 817, in parse_db
    entries.append( data )
UnboundLocalError: local variable 'data' referenced before assignment
```

I did a bit of archaeology and this seems to be there since a long time.

ErikBjorge, 2018-04-16:
This may be due to chipsec-util using the Linux UEFI Variable data in /sys/firmware/efi/efivars/.  We can force it to use the driver and see if that is a workaround for this issue.  Do you mind updating the requires_driver function in chipsec/utilcmd/uefi_cmd.py.  Just add 'var-list' to the ops that should load the chipsec driver.

return (op in ('tables','s3bootscript','var-list'))

ErikBjorge, 2018-04-16:
Ignore my previous suggestion.  Adding the one line will not work around the issue.  We will continue to investigate the problem from our side.  We may need some additional information if we can't reproduce the failure on our systems here.

martingalvan, 2018-04-16:
Sure, just let me know.

ErikBjorge, 2018-04-16:
Can you add the following print to chipsec/hal/uefi.py at line 229.  Then rerun the command and you should know what variable is causing the parsing problem.  It would also be helpful if you would be willing to send the failing variable file so we can check the formatting.

```python
  var_fname = os.path.join( nvram_pth, '%s_%s_%s_%d.bin' % (name, guid, attr_str.strip(), n) )
  write_file( var_fname, data )
+ print('- Variable File: {}'.format(var_fname))
  #if name in SECURE_BOOT_VARIABLES:
  if name in AUTHENTICATED_VARIABLES:
      parse_efivar_file( var_fname, data )
```

I have a potential fix but want to understand the failure mode more before I make a decision.  The failing variable data will help me do that.

martingalvan, 2018-04-16:
Sure. The last variable whose name appears before the exception is efi_variables.dir/certdb_d9bee56e-75dc-49d9-b4d7-b534210f637a_NV+BS+RT+TBAWS_0.bin, I'm attaching it [here](https://github.com/chipsec/chipsec/files/1917279/certdb_d9bee56e-75dc-49d9-b4d7-b534210f637a_NV.BS.RT.TBAWS_0.zip).

ErikBjorge, 2018-04-16:
Thanks for sharing.  I will look at this now and see if I can fix the issue.

ErikBjorge, 2018-04-16:
After looking at the variable you sent I found the issue.  It looks like your system has uncovered at least three problems parsing authenticated variables.

1. Chipsec does not handle the certdb/certdbv variables correctly.  They have a different format than the secure boot variables.
1. Chipsec does not have all the current signature/cert GUID information.
1. The error case code you hit is broken and needs to be fixed/removed.  It should just display an error and bail.

We will be working on fixing these issues.

martingalvan, 2018-04-17:
Sounds great, thanks Erik!

ErikBjorge, 2018-05-16:
Would you be willing to test a set of changes to fix the issue?  I have it in my fork of chipsec on the dec/cert-db-processing branch.  Could you pull my repo and see if the changes on the branch solve this issue?

https://github.com/ErikBjorge/chipsec

ErikBjorge, 2018-05-16:
Let me improve the warning message so I can determine what variable is actually failing.  I will update the fix and would appreciate if you would test again.

Were you able to find the cert information from the certdb variable?

martingalvan, 2018-05-16:
I don't know :) where should it be stored?

ErikBjorge, 2018-05-16:
Pushed up a couple more patches to the sane branch that should help debug the issue.  I think you may have a AuthVarKeyDatabase variable that I have not had time to add parsing for.  If you have time to run another test please send me the efi_variables.lst file and the AuthVarKeyDatabase_*.bin file if it exists.

martingalvan, 2018-05-23:
Hi Eric! Sorry for the delay. Here are the files:

[efivars.zip](https://github.com/chipsec/chipsec/files/2032457/efivars.zip)

I noticed that the AuthVarKeyDatabase...bin file was placed at the root of efi_variables.dir. That dir in turn has an empty AuthVarKeyDatabase...dir directory within it.

ErikBjorge, 2018-05-23:
Not a problem.  Will see if I can add the parsing for this one as well.

ErikBjorge, 2018-06-07:
Added parsing for this as well.  This variable was a little odd in how it is implemented.  If the variable is not present the variable will be created with a single byte of data (0).  I am submitting the pull request today for additional testing.

ErikBjorge, 2018-06-20:
PR #400 has been merged.  If this did not resolve the issue please reopen this issue.

closed 2018-06-20 by ErikBjorge