chipsec/chipsec #375: PREFast Defect Log and misc items

DropBearLabs, 2018-04-19:
Who's the appropriate contact to discuss bits around the win7 driver for Chipsec pertaining to:

- PREFast defect log
- paths to move away from test-signing hack
- transpiling chipsec with say Nuitka vs current py2exe ( which also has issues )

DropBearWare, 2018-04-19:
@johnloucaides -  It's Arun Koshy here from the mail-thread from Ron a week or so ago.

DropBearWare, 2018-04-19:
@ErikBjorge -- happy to help @BrentHoltsclaw if he needs any more info though I suspect he'll probably be able to repro without much drama.

Thanks for clarifying the signing posture.. makes sense. Re transpiling, will email after a few more experiments.

Huge props to the team for building this ! Thanks to all involved for their hard work !

ErikBjorge, 2018-12-21:
Closing unless additional support is needed.

closed 2018-12-21 by ErikBjorge