chipsec/chipsec #377: No output when execute chpsec_util.main(['uefi', 'tables'])

rhapsodyjay, 2018-04-26:
[image]
I get above results but found nothing generated in my Chipsec folder.
Anything missed??

ErikBjorge, 2018-04-26:
I will take a look at this today and provide any feedback on this issue.

ErikBjorge, 2018-04-26:
I did a quick test of this and did not see an issue.  It looks like chipsec is not finding any of the UEFI tables and that explains the missing output.

Is this running in a VM?  Can you provide me with more system configuration data?

rhapsodyjay, 2018-04-27:
No, it runs on Win10 x64 RS3 on Intel CFL-S chipsets with secure boot disabled.

rhapsodyjay, 2018-05-15:
Hi Erik,

I dumped the info as below.
[tables.txt](https://github.com/chipsec/chipsec/files/2003069/tables.txt)

ErikBjorge, 2018-05-15:
So that is very interesting.  It looks like the script found the IBI SYST signature but the field verification did not succeed.  It looks like we just don't have the 2.6 version information.  I will submit a patch for this today.

ErikBjorge, 2018-05-15:
Please check to see if change #385 fixes the problem.

ErikBjorge, 2018-05-16:
Thanks for testing.  We will get this committed soon.

ErikBjorge, 2018-05-17:
Change merged.  Please reopen the issue if this issue shows up again.

closed 2018-05-17 by ErikBjorge