chipsec/chipsec #378: chipsec_util.main(['spi', 'dump', 'rom.bin']) dumps 16MB bios rom file, but my bios is 32MB.

rhapsodyjay, 2018-04-26:
[image]
The dumped rom.bin is  only 16MB.
[image]
But my BIOS image is actually 32MB(16MB bios + 16MB ME).

ErikBjorge, 2018-04-26:
After looking at spi_cmd.py it looks like the dump command it looks like it may have a few bugs that need to be fixed.  Looks like it may have some assumptions that are no longer valid.  Please try using the spi read command and provide the start offset and length and file for output.

chipsec_util spi read 0x0 0x2000000 bios.bin

rhapsodyjay, 2018-04-27:
Thanks!
I've tried chipsec_util spi read 0x0 0x2000000 bios.bin.
It works fine.

ErikBjorge, 2018-05-14:
Thanks for the pointer.  This is an option but comes from the flash descriptor.  I will see if I can extract this information from the SPI controller instead of reading the flash descriptor directly (don't want two flash reads).  I think I can get the data via the BIOS_FDOC and BIOS_FDOD registers.

I think the real issue for me is the intent of the dump command.  Was the intent to dump only the BIOS region or the entire flash device?  The code specifically attempts to determine the size of the BIOS region and dump it.  This conflicts with the message at the start that says that it dumps the entire flash device.

This leaves us with two options. We could just fix get_SPI_region to correctly determine the BIOS region or we could update dump to determine the size of the flash part.  I am not sure of the best option.

skochinsky, 2018-05-15:
IMO the default "dump" command should try to dump the whole chip. I imagine you could eventually add FPT-like options to dump specific regions.

fproj, 2018-05-15:
agree with @skochinsky

ErikBjorge, 2018-05-15:
I am fine with dumping the entire flash image with the dump command.  This does seem the most reasonable.  I think we should also add a dump-bios command that will just dump the BIOS region.  This way we will also have the legacy functionality available.

abazhaniuk, 2018-05-18:
BIOS_FDOC and BIOS_FDOD registers not returning you size of SPI chipsec, is it? Not sure how it will help to identify size of SPI chip...

tweksteen, 2018-06-11:
I believe the intent has always been to dump the whole flash. The BIOS region address was used under the (fair) assumption that its end will correspond to the end of the flash. I can't think of any exception to this?

To the original question, I've come across this issue multiple time with the incorrect definition of the PCH. Could you run `chipsec_util platform`? If you see the "Default PCH" being used, it is likely that your chipset is unknown. You could: find the correct PCI DID, the matching datasheet and add an entry for the PCH version. You can also test using the "--pch" flag of chipsec_util.

rhapsodyjay, 2018-08-02:
Can read the JEDEC ID by 
[image]
But this way needs to setup a table to refer to.

closed 2026-10-09 by npmitche