chipsec/chipsec #381: Python 2 EOL

DropBearWare, 2018-05-01:
Chipsec seems to be done in Python 2 which enters official end of life phase on Jan 1st 2020. Time remaining now to port is < two years which probably is a good bit to keep in mind for a project of this complexity.

We could get started on an early analysis of what it would take to upgrade to Python 3.

-- References --

- https://github.com/python/devguide/pull/344

- https://pythonclock.org

ErikBjorge, 2018-05-01:
Yes, this is on our list to do this year.  The current plan is to enable Python 2/3 compatibility so we don't break people too bad.  Once we have Python 3 support released we can define a date where we may not support Python 2 failures in CHIPSEC.

Our current plan is to start enabling chipsec_util with the mixed python support and thus get to the HAL/Helper modules as well.  Once we know how much fun it will be from enabling chipsec_util we can complete the chipsec_main support conversion and module updates.

To support this development we should also create a shared development branch.

If you have comments or suggestions please let us know.

DropBearWare, 2018-05-02:
Glad to assist with testing, dev as this develops. Is the mentioned TO-DO for the year public or internal? If public, could you please share the link ?

ErikBjorge, 2018-05-02:
We have just started scoping the task.  Once we have a plan we will post it on the wiki for all to see and track progress.

ErikBjorge, 2018-07-24:
Added Wiki page and branch for this task.
https://github.com/chipsec/chipsec/wiki/Python-3-Conversion

hughsie, 2019-03-27:
If it's a datapoint, I can't use CHIPSEC in the LVFS until it can run on Python3.

BrentHoltsclaw, 2019-03-27:
The python-3 beta branch is almost feature complete at this point.  There are issues with the compress and decompress functionality which will be addressed soon.  We are going to merge the python 3 branch with the main branch again soon.  After that the focus will be on fixing up the unit testing.

hughsie, 2019-05-10:
> We are going to merge the python 3 branch with the main branch again soon

Do you have a rough ETA? I'm being pushed for a feature and I don't know whether to use `chipsec` or `uefi-firmware-parser` for decompression. I'd obviously like to use the former if possible :)

BrentHoltsclaw, 2019-05-13:
All necessary pull requests have been merged into master.  I'm going to create a python3 rc1 branch and release by tomorrow.  There is still an issue with signed files that I need to put in a pull request for, but will work to get that updated by the end of the week.

kuqin12, 2019-05-22:
Hi may i know whether py3 change is in the feature branch yet? I am interested in this feature :) Thanks

BrentHoltsclaw, 2019-05-22:
We are working to get python3 RC-1 release out on 5/24.

BrentHoltsclaw, 2019-05-24:
python3-rc1 has been released with linux support.  We will be working to enable other major OS support in the coming weeks.  Please let us know if you run into any issues.

BrentHoltsclaw, 2019-10-10:
Python3 support has been added.  Please open any problems with the python3 implementation as new issues.  We will open a new issue for UEFI shell support.

closed 2019-10-10 by BrentHoltsclaw