chipsec/chipsec #395: Core dump in chipset_main

jurekb, 2018-05-20:
Arch Linux, AUR repo:
chipsec-dkms-git v1.2.5.r20.gc377a72-1
chipsec-git  v1.2.5.r20.gc377a72-1

# /usr/bin/chipsec_main -l chipset_main.log
$ cat chipset_main.log
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] Arguments: -l chipset_main.log
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x2A00
ERROR: Platform is not supported (Unsupported Platform: VID = 0x8086, DID = 0x2A00).
ERROR: To run anyways please use -i command-line option

# /usr/bin/chipsec_main -i -l chipset_main-i.log
[*] Ignoring unsupported platform warning and continue execution
Segmentation fault
$ cat chipset_main-i.log
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] Arguments: -i -l chipset_main-i.log
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x2A00
ERROR: Platform is not supported (Unsupported Platform: VID = 0x8086, DID = 0x2A00).
WARNING: Platform dependent functionality is likely to be incorrect
[CHIPSEC] OS      : Linux 4.16.9-1-ARCH #1 SMP PREEMPT Thu May 17 02:10:09 UTC 2018 x86_64
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2A00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2815
 
[*] loading common modules from "/usr/lib/python2.7/site-packages/chipsec/modules/common" ..
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.memlock
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.cpu.spectre_v2
[+] loaded chipsec.modules.common.uefi.s3bootscript
[+] loaded chipsec.modules.common.uefi.access_uefispec
[*] No platform specific modules to load
[*] loading modules from "/usr/lib/python2.7/site-packages/chipsec/modules" ..
[+] loaded chipsec.modules.smm_dma
[+] loaded chipsec.modules.remap
[+] loaded chipsec.modules.debugenabled
[+] loaded chipsec.modules.memconfig
[*] running loaded modules ..

[*] running module: chipsec.modules.common.rtclock
Skipping module chipsec.modules.common.rtclock since it is not supported in this platform

[*] running module: chipsec.modules.common.spi_desc
[x][ =======================================================================
[x][ Module: SPI Flash Region Access Control
[x][ =======================================================================
[*] FRAP = 0x00000000 << SPI Flash Regions Access Permissions Register (SPIBAR + 0x50)
    [00] BRRA             = 0 << BIOS Region Read Access 
    [08] BRWA             = 0 << BIOS Region Write Access 
    [16] BMRAG            = 0 << BIOS Master Read Access Grant 
    [24] BMWAG            = 0 << BIOS Master Write Access Grant 
[*] Software access to SPI flash regions: read = 0x00, write = 0x00

[+] PASSED: SPI flash permissions prevent SW from writing to flash descriptor

[*] running module: chipsec.modules.common.ia32cfg
[x][ =======================================================================
[x][ Module: IA32 Feature Control Lock
[x][ =======================================================================
[*] Verifying IA32_Feature_Control MSR is locked on all logical CPUs..
[*] cpu0: IA32_Feature_Control Lock = 1
[*] cpu1: IA32_Feature_Control Lock = 1
[+] PASSED: IA32_FEATURE_CONTROL MSR is locked on all logical CPUs

[*] running module: chipsec.modules.common.bios_wp
[x][ =======================================================================
[x][ Module: BIOS Region Write Protection
[x][ =======================================================================
[*] BC = 0x01 << BIOS Control (b:d.f 00:31.0 + 0xDC)
    [00] BIOSWE           = 1 << BIOS Write Enable 
    [01] BLE              = 0 << BIOS Lock Enable 
    [02] SRC              = 0 << SPI Read Configuration 
    [04] TSS              = 0 << Top Swap Status 
    [05] SMM_BWP          = 0 << SMM BIOS Write Protection 
[-] BIOS region write protection is disabled!

[*] 
#

$ journalctl
May 20 09:50:47 myhost kernel: Chipsec module loaded 
May 20 09:50:47 myhost kernel: ** This module exposes hardware & memory access,>
May 20 09:50:47 myhost kernel: ** which can effect the secure operation of     >
May 20 09:50:47 myhost kernel: ** production systems!! Use for research only!  >
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > IOCTL_RDMMIO
May 20 09:50:47 myhost kernel: [chipsec] > READMEM
May 20 09:50:47 myhost kernel: [chipsec] > READMEM
May 20 09:50:47 myhost kernel: [chipsec] > READMEM
May 20 09:50:47 myhost kernel: general protection fault: 0000 [#1] PREEMPT SMP >
May 20 09:50:47 myhost kernel: Modules linked in: chipsec(O) ccm fuse ip6t_REJE>
May 20 09:50:47 myhost kernel:  soundcore tg3 ptp led_class intel_gtt pps_core >
May 20 09:50:47 myhost kernel: CPU: 0 PID: 8436 Comm: chipsec_main Tainted: G  >
May 20 09:50:47 myhost kernel: Hardware name: LENOVO 7650E6G/7650E6G, BIOS 7PET>
May 20 09:50:47 myhost kernel: RIP: 0010:_rdmsr+0xf/0x1e [chipsec]
May 20 09:50:47 myhost kernel: RSP: 0018:ffffa46e48acfd38 EFLAGS: 00010246
May 20 09:50:47 myhost kernel: RAX: ffffa46e48acfdb0 RBX: 00007fff3f60b8e0 RCX:>
May 20 09:50:47 myhost kernel: RDX: ffffa46e48acfdc0 RSI: ffffa46e48acfdc8 RDI:>
May 20 09:50:47 myhost kernel: RBP: ffffa46e48acfe70 R08: 000000000000003a R09:>
May 20 09:50:47 myhost kernel: R10: ffffa46e48acfdc8 R11: ffffa46e48acfdc0 R12:>
May 20 09:50:47 myhost kernel: R13: 00000000c0080005 R14: 00007fff3f60b8e0 R15:>
May 20 09:50:47 myhost kernel: FS:  00007fc95bd234c0(0000) GS:ffff8eab7f400000(>
May 20 09:50:47 myhost kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May 20 09:50:47 myhost kernel: CR2: 00007fc95b51adc0 CR3: 00000000605c4000 CR4:>
May 20 09:50:47 myhost kernel: Call Trace:
May 20 09:50:47 myhost kernel:  ? valid_mmap_phys_addr_range+0x656/0xa80 [chips>
May 20 09:50:47 myhost kernel:  ? _raw_spin_unlock+0x16/0x30
May 20 09:50:47 myhost kernel:  ? do_vfs_ioctl+0xa4/0x610
May 20 09:50:47 myhost kernel:  ? do_vfs_ioctl+0xa4/0x610
May 20 09:50:47 myhost kernel:  ? __do_page_fault+0x237/0x570
May 20 09:50:47 myhost kernel:  ? SyS_ioctl+0x74/0x80
May 20 09:50:47 myhost kernel:  ? do_syscall_64+0x74/0x190
May 20 09:50:47 myhost kernel:  ? entry_SYSCALL_64_after_hwframe+0x3d/0xa2
May 20 09:50:47 myhost kernel: Code: 0f 01 0f c3 0f 01 1f c3 0f 01 07 c3 0f 01 >
May 20 09:50:47 myhost kernel: RIP: _rdmsr+0xf/0x1e [chipsec] RSP: ffffa46e48ac>
May 20 09:50:47 myhost kernel: ---[ end trace 3d87bbae6302c0f0 ]---
$

jurekb, 2018-05-20:
[debug.log](https://github.com/chipsec/chipsec/files/2020543/debug.log)
[dmesg.log](https://github.com/chipsec/chipsec/files/2020544/dmesg.log)

ErikBjorge, 2018-05-21:
The reason for the unsupported platform message is that we tend not to support older platforms prior to the release of CHIPSEC.  It looks like this is a 965 platform and we do not have plans to add support at this time.  If someone submits a configuration file for this chipset we would be willing to add it to the tool.

Also note that running chipsec_main with the -i command line option will not provide reliable results as appropriate register definitions will not be loaded by the tool.  This can lead to invalid results and crashes.

We are looking at the error you encountered as it is not clear why the failure is happening.  Based on the test output it should be doing an MMIO read of the SPI controller.  When we look at the dmesg output it looks like an MSR read is occurring.  We will continue to look at this as you may have found an interesting bug in our driver code.

We do have one question.  Did you run dmesg immediately after running chipsec to collect the log data?  It looks like you did but we just wanted to check to make sure they are a matched set of logs.

jurekb, 2018-05-21:
Actually I built chipsec from the latest source (the package version descriptions in Arch Linux AUR repo were fixed, old and incorrect). I corrected the original ticket.

As requested I also obtained very latest source and tried to build chipsec:
[jb@myhost chipsec-current]$ git describe --long --tag
v1.3.6rc1-19-ged76b7b

I could not get a clean compile run with python3:
$ python -V
Python 3.6.5
[jb@myhost chipsec-current]$ python setup.py build_ext -i
$
[chipset-bad-compile.log](https://github.com/chipsec/chipsec/files/2023963/chipset-bad-compile.log)

I succeeded with python2:
$ python2 -V
Python 2.7.15
[jb@myhost chipsec-current]$ python2 setup.py build_ext -i
[jb@myhost chipsec-current]$ sudo -u root python2 chipsec_main.py -i -d -v -l new-driver.log
[*] Ignoring unsupported platform warning and continue execution
Segmentation fault
$
[new-driver.log](https://github.com/chipsec/chipsec/files/2023970/new-driver.log)
[dmesg-new-driver.log](https://github.com/chipsec/chipsec/files/2023971/dmesg-new-driver.log)

ErikBjorge, 2018-05-21:
We will look at the log information and see if we can debug the issue.

Currently we only support Python 2.7.x.  We are working on enabling Python 3.6 support in the future.

BrentHoltsclaw, 2018-05-21:
@jurekb I have modified our driver to give additional output to help with debugging the issue.  Could you please replace the linux driver with the included file and recompile as above.  Afterwards could you run chipsec in two ways
`sudo python chipsec_main.py -i -d -v -l dbg-driver.log`
`sudo python chipsec_main.py -i -d -v -l dbg-bioswp.log -m common.bios_wp`

[chipsec_km.zip](https://github.com/chipsec/chipsec/files/2024330/chipsec_km.zip)

jurekb, 2018-05-22:
Compilation:
[compile-with-chipsec_km.log](https://github.com/chipsec/chipsec/files/2025087/compile-with-chipsec_km.log)

[jb@myhost chipsec-current]$ sudo -u root python2 chipsec_main.py -i -d -v -l dbg-driver.log
[sudo] password for root: 
[*] Ignoring unsupported platform warning and continue execution
Segmentation fault
$
[dbg-driver.log](https://github.com/chipsec/chipsec/files/2025095/dbg-driver.log)
[dmesg-dbg-driver.log](https://github.com/chipsec/chipsec/files/2025097/dmesg-dbg-driver.log)

[jb@myhost chipsec-current]$ sudo -u root python2 chipsec_main.py -i -d -v -l dbg-bioswp.log -m common.bios_wp
[sudo] password for root: 
[*] Ignoring unsupported platform warning and continue execution
$
[dbg-bioswp.log](https://github.com/chipsec/chipsec/files/2025112/dbg-bioswp.log)
[dmesg-dbg-bioswp.log](https://github.com/chipsec/chipsec/files/2025117/dmesg-dbg-bioswp.log)

jurekb, 2018-05-22:
@BrentHoltsclaw 
"f you would like to run a full run, either disable the test by deleting it from the directory ..."
Do you mean deleting tests/software/test_msr.py ? Or some other test ?

ErikBjorge, 2018-05-22:
Because your platform is not supported by chipsec the results may not be accurate.  The use of the -i flag is really only for running specific tools in unsupported environments and not for general testing.  In many cases the tests may be reading incorrect registers or checking for features that were not yet available in that processor/chipset generation.

An in depth investigation of the results would be required to determine what results could be trusted and what results can't.

ErikBjorge, 2018-12-21:
Closing issue unless further support is required.

closed 2018-12-21 by ErikBjorge