chipsec/chipsec #398: Chipsec doesn't decode Dell R620 UEFI firmware image correctly

wrffrz, 2018-06-04:
I tried using chipsec to decode a SPI dump from the latest version of firmware from a Dell R620 (Sandy Bridge), but it does not find anything in the image:

	$ git pull
	Already up-to-date.
	$ python chipsec_util.py uefi decode ../dell-r620.bin

	################################################################
	##                                                            ##
	##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
	##                                                            ##
	################################################################
	[CHIPSEC] Version 1.3.6rc1
	[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
	[CHIPSEC] Executing command 'uefi' with args ['decode', '../dell-r620.bin']

	[CHIPSEC] Parsing EFI volumes from '../dell-r620.bin'..
	[CHIPSEC] (uefi) time elapsed 0.372

	$ ls -laR ../dell-r620.bin.dir/
	../dell-r620.bin.dir/:
	total 12
	drwxrwxr-x 3 jesse jesse 4096 Jun  4 12:51 .
	drwx------ 5 jesse jesse 4096 Jun  4 12:51 ..
	drwxrwxr-x 2 jesse jesse 4096 Jun  4 12:51 FV

	../dell-r620.bin.dir/FV:
	total 8
	drwxrwxr-x 2 jesse jesse 4096 Jun  4 12:51 .
	drwxrwxr-x 3 jesse jesse 4096 Jun  4 12:51 ..
	$

This is a the latest version of the UEFI firmware for this platform which was released March 17, 2018 and is available here: https://www.dell.com/support/home/us/en/04/Drivers/DriversDetails?driverId=94DFH

BrentHoltsclaw, 2018-06-05:
Chipsec decode of FV does not currently follow UEFI specifications.  This will require a rewrite of the decode functionality.

wrffrz, 2018-06-06:
I took a closer look at the Dell R620 firmware and although it's a UEFI firmware, it doesn't have the _FVH firmware volume signature anywhere in the image and I don't see any of the EFI_FIRMWARE_FILE_SYSTEM{,2,3}_GUID bytes anywhere either.

ErikBjorge, 2018-06-06:
What file did you download and what process did you use to extract the actual BIOS image?  In most cases it looks like the bios/capsule image is wrapped by an executable or a shell script.

wrffrz, 2018-06-06:
I originally dumped the BIOS image by using chipsec_util spi dump and also was able extract an image from the following download package by using 7z to extract the file contents:

File Format:Update Package for Microsoft® Windows® 64-Bit.
File Name:BIOS_94DFH_WN64_2.6.1.EXE

The payload/R620-020601C.hdr file has a 84-byte header and a 264-byte footer that should be removed, but once those are removed the beginning and end of this image matches the contents extracted from the live system using chipsec_util spi dump.

  7z x BIOS_94DFH_WN64_2.6.1.EXE
  cd payload
  dd if=R620-020601C.hdr of=r620.bin bs=1 skip=84 count=8388608

skochinsky, 2018-06-07:
This BIOS is legacy 16/32-bit code, pre-UEFI

ErikBjorge, 2018-12-21:
Closing issue unless additional help is required.

closed 2018-12-21 by ErikBjorge