chipsec/chipsec #399: Chipsec doesn't decode Acer KBL firmware image

abazhaniuk, 2018-06-06:
[firmware_acer.zip](https://github.com/chipsec/chipsec/files/2078160/firmware_acer.zip)
It is firmware from Acer Predator GX-792

chipsec extracted 66 DXE/PEI files. 
uefi-firmware-parser ~300 
I think some FV was not decoded, maybe related to https://github.com/chipsec/chipsec/issues/367

ErikBjorge, 2018-06-06:
Seems like it may be the same as #398 as well.

skochinsky, 2018-06-07:
My guess would be the LZMA-compressed nested volume.

BrentHoltsclaw, 2018-06-12:
Can you try pulling https://github.com/brentHoltsclaw/chipsec/tree/uefi_decompress and see if it fixes your issue?

BrentHoltsclaw, 2018-06-20:
@abazhaniuk @johnloucaides Did you get a chance to verify whether the decode functionality fixes the issue and that it does not cause any other issues?

ErikBjorge, 2018-12-21:
This should be fixed in the latest release using the updated decompression code.  Please reopen if the failure continues.

closed 2018-12-21 by ErikBjorge