chipsec/chipsec #401: Add support for C620 series

tweksteen, 2018-06-08:
This should be quite straightforward to add the C621-C628 chipsets. I am not sure if this should be added as PCH 200 or under its own definition.

ErikBjorge, 2018-06-08:
Let me check the specifications and see if they are from the same generation.  It may be that we can use the 200 series PCH configuration files.  Just need to double check a few things to see if they diverge too much.

ErikBjorge, 2018-06-11:
For now we can either use the 100 or 200 series PCH configuration file.  If we need access to some of the other C620 devices (such as IE HECI) we may need to create an individual configuration file.  I would just use the 200 series PCH configuration for now.

tweksteen, 2018-06-12:
Ack. I went down the path of adding a new file, mainly to expose the extra flash regions. See PR #404. It is based on the 200 series definition.

closed 2018-06-29 by tweksteen