chipsec/chipsec #406: Chipset is able to reflash bios at boot

ZandroidX, 2018-06-19:
I recently reflashed my bios because I noticed it was infected. I pulled the image from the chip and checksum diffed the stock, and pulled bios image. They did not match up, thinking the only solution would be to reflash and no problem right? Wrong. I reflashed and rebooted without booting into an os and my bios was instantly reflashed with an infected image. How can the chipset do this without having an operating system environment to reinect it?

ErikBjorge, 2018-06-19:
How did you determine that your BIOS was infected?

Just verifying the checksum of the flash part is not a good method for detecting infection or even correctness.  Some regions of the flash will be modified during the boot process and this will obviously change the checksum.  These modifications are expected as firmware stores settings into non-volatile storage for use on future boots.  In most cases these modifications take the form of UEFI Variables.

One option to check your BIOS would be to use the whitelist functionality to generate a known good whitelist from the source firmware image and then extract and check your system image.

ErikBjorge, 2018-12-21:
Closing issue.  Please reopen if additional support is needed.

closed 2018-12-21 by ErikBjorge

ZandroidX, 2018-12-23:
I determined it was infected because when checked the hash of the infected chip, flashed the chip externally and put it back in the computer, booted it up turned off the computer checked the hash again and it was not the same checksum as the stock bios image.

sskras, 2018-12-23:
@ZandroidX , can you confirm what @ErikBjorge says?

> Some regions of the flash will be modified during the boot process and this will obviously change the checksum. These modifications are expected as firmware stores settings into non-volatile storage for use on future boots. 

@ErikBjorge , I am an oldschool guy who learned PC arch circa 1995, when non-volatile storage used by BIOS was implemented as a separate CMOS memory chip (which was kept powered from a battery on mainboard). No wonder since BIOS was a real ROM (or UV-EPROM, IIRC) at the time, not a Flash-based EEPROM. As the battery is still present on mainboads, I am intrigued about what has changed over time:

1. Does the BIOS / Firmware store settings in two places?
2. Does it differentiate which settings should be written into CMOS chip and which into Flash chip?
3. Why are then the batteries on mainboards still used?
4. Is CMOS content mirrored to the Flash during power on?
5. Is Flash non-volatile store overwritten (reset) when CMOS battery gets disconnected, connected and then system powers on?
6. Or are these two mirrored in a different way?

ZandroidX, 2018-12-24:
Whenever I re-install an operating system, Before the installer screen shows a command prompt window shows up then quickly disappears. I know that there is some pre-boot malware hiding somewhere in my system.

ZandroidX, 2019-01-02:
My Root Complex Base Address Register (RCRB) has been relocated and now I cannot find My General Control Status register located at offset of 3410 from RCRB. It looks like this malware has completely screwed with my base address registers and now I don't know where anything is at.

ZandroidX, 2019-01-02:
I cannot pull my bios image off this machine as it has an EC (Embedded Controller) which prevents me from flashing or reading my BIOS. I scrapped my other machine since the original post on this.

ErikBjorge, 2019-01-02:
What processor and chipset are you using?  Can you send a PCI configuration space dump so we can check the base register and verify it against expected values?  It is difficult to determine what might be wrong without being able to see what changed and the value used.

ErikBjorge, 2019-01-02:
You should not have to wait a long time for the file write to complete.  Based on the information you sent the base address is actually FED1C000 because the low bit is the enable bit.  You should be reading address FED1F410.  The address is programmed by firmware on boot.

ZandroidX, 2019-01-02:
I updated my comment, I tried that address first with no luck.

ErikBjorge, 2019-01-03:
The value FED1C000 is normally the expected address.  What value do you get when you read the register?  Also, are you attempting to change bits 11:10?

ZandroidX, 2019-01-03:
it doesn't allow me to change the bits and just sets to 00000801 when attempting to change bits 11:10

ZandroidX, 2019-01-03:
Is there any way to access the PCI Boot so I can reset my base address registers, since I am now unable to boot from my SPI device like I'm supposed to?

ErikBjorge, 2019-01-03:
You still may be booting off of SPI regardless of the GCS Boot BIOS Straps.  I need to do some more checking on this.  Settings in the flash descriptor may change behavior of these straps or ignore them completely.

Please double check that you are accessing the correct MMIO address to modify the GCS.

ZandroidX, 2019-01-03:
Yeah this is not the GCS register, when I reboot the machine bits 11:10 are reset to 0.