chipsec/chipsec #408: chipsec-1.3.6rc1 not be compatible for purley platform

zhaomy163, 2018-06-21:
Hi Chipsec team:

Get following message when try to run Chipsec on Purley platform;
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] Arguments: -m common.spi_access
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x2020
ERROR: Platform is not supported (Unsupported Platform: VID = 0x8086, DID = 0x2020).
ERROR: To run anyways please use -i command-line option

Get some failed items if run with -i parameter,but which is passed according to bios post debug message;
Check the source code,the C621 PCH and skylake server CPU ID was not included;

Try to update latest version but no improvement.
If we can refer to the result with -I parameter for Purley platform?

BR!
Meiyu

ErikBjorge, 2018-06-21:
The Purley platform support has not been released yet.  I will assign someone to work on getting the platform released.

I would not use the -i parameter for general testing because the correct register definitions will not be used.  This will just produce invalid test results.  The -i parameter should only be used in some limited cases when testing in a VM or running specific tools.

zhaomy163, 2018-06-26:
Hi ErikBjorge:
Thanks for your help,we will test until chipsec update to support purley.
Also will pass related information to our customers.
BR!
Meiyu

gengcs, 2018-06-26:
Hi ErikBjorge:

I find chipsec has a parameter -p SKL, In my mind, this parameter is for platform. could I use the parameter to scan purley platform? or use -i and -p SKL to scan purley platform?

thanks.
GengCS.

ErikBjorge, 2018-06-27:
That will not work as expected because it will load the desktop configuration not the server specific configuration file.  This will generate incorrect results.  In most cases the server configuration will end with an 'X'.  See the BDW and BDX configurations as examples of the differences.

elicn, 2019-01-07:
As part of the Purely platform compatibility, the PCI bus numbering should be fixed: while devices on desktops are assigned a constant bus number, in servers it is different - and CHIPSEC dopesn't support that.
For example, the notation B(0)/D31/F0 means the device will be found on a bus number which was allocated to B0; it could be 0, but it's not necessarily the case. To find out what which bus number was actually assigned to B(0-5), CHIPSEC should consult MSR 300h [notice this MSR is in package scope, so different cores / threads may address different bus numbers to reach the same device].

ErikBjorge, 2019-01-07:
Yes, this is a problem with chipsec and needs to be resolved for Xeon based systems.  Thanks @elicn for pointing this out.  I will file an issue for this so we can track it independently of Purley support.  Please feel free to contribute to the new issue with any information that will help us resolve the problem.

npmitche, 2024-02-10:
I believe this was resolved with PR https://github.com/chipsec/chipsec/pull/580

closed 2024-02-10 by npmitche