chipsec/chipsec #410: ioctl string arg too long

abazhaniuk, 2018-06-21:
In latest version of chipsec we have:
```
[CHIPSEC] Dumping IDT of 4 CPU threads
[cpu0] IDT:
Traceback (most recent call last):
  File "chipsec_util.py", line 268, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 264, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 231, in main
    comm.run()
  File "/home/virvdova/chipsec/git/chipsec/chipsec/utilcmd/desc_cmd.py", line 47, in run
    self.cs.msr.IDT_all( 4 )
  File "/home/virvdova/chipsec/git/chipsec/chipsec/hal/msr.py", line 164, in IDT_all
    self.IDT( tid, num_entries )
  File "/home/virvdova/chipsec/git/chipsec/chipsec/hal/msr.py", line 157, in IDT
    return self.dump_Descriptor_Table( cpu_thread_id, DESCRIPTOR_TABLE_CODE_IDTR, num_entries )
  File "/home/virvdova/chipsec/git/chipsec/chipsec/hal/msr.py", line 136, in dump_Descriptor_Table
    dt = self.helper.read_physical_mem( pa, limit + 1 )
  File "/home/virvdova/chipsec/git/chipsec/chipsec/helper/oshelper.py", line 222, in read_physical_mem
    return self.helper.read_phys_mem( (phys_address>>32)&0xFFFFFFFF, phys_address&0xFFFFFFFF, length )
  File "/home/virvdova/chipsec/git/chipsec/chipsec/helper/linux/helper.py", line 364, in read_phys_mem
    out_buf = self.ioctl(IOCTL_READ_PHYSMEM, in_buf)
  File "/home/virvdova/chipsec/git/chipsec/chipsec/helper/linux/helper.py", line 304, in ioctl
    return fcntl.ioctl(self.dev_fh, self._ioctl_base + nr, args)
ValueError: ioctl string arg too long
```
in many interfaces. for read memory, idt, gdt, etc.

ErikBjorge, 2018-06-22:
We have a fix for read/write of memory.  Will look at IDT, GDT and other interfaces.

ErikBjorge, 2018-06-29:
As I dug into this I have found another issue in the Linux driver that I am currently working on fixing.  Sorry for the delay getting this fixed.

ErikBjorge, 2018-06-29:
I was able to fix the general read/write of physical memory but when working on the GDT/IDT/LDT command I found another issue.  It seems like we have an issue in the my_xlate_dev_mem_ptr and the fact that guess_page_is_ram seems to be returning FALSE in most cases even when I know the address is memory.

I am going to do some more testing on the memory read/write update and submit a patch.  After that I will do some more debug on why we are not correctly converting the PA to a VA.

ErikBjorge, 2018-07-03:
These changes should fix the basic read/write changes.  I still have not found a good solution to the IDT/GDT/LDT memory mapping issue.  Here is my branch with the current patch set while I work on the other issues.

https://github.com/ErikBjorge/chipsec/tree/dev/read-write-phys-mem

ErikBjorge, 2018-09-17:
Yes, I will submit a partial fix for the issue this week.  This fixes the ioctl issue but other problems with memory access will still exist.

ErikBjorge, 2018-09-25:
I think the solution is to revert to the old implementation.  Even with reverting to the older implementation I am seeing issues with dumping the GDT/IDT.  It seems the address being requested is outside the physical memory range.  Still need to debug why that is happening.

ErikBjorge, 2018-10-24:
Reverted back to original implementation.

ErikBjorge, 2018-12-21:
Closing issue until we break things again...

closed 2018-12-21 by ErikBjorge