chipsec/chipsec #413: Kernel module does not load in Fedora  4.16.14-200.fc27

ismaws, 2018-07-04:
Related to: https://github.com/chipsec/chipsec/issues/363

Hitting:

Build:
WARNING: "valid_mmap_phys_addr_range" [/home/jirangel/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!

Install:
copying drivers/linux/amd64/cpu.asm -> /home/jirangel/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
WARNING: "valid_mmap_phys_addr_range" [/home/jirangel/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!

dmesg:
[    0.000000] Linux version 4.16.14-200.fc27.x86_64 (mockbuild@bkernel02.phx2.fedoraproject.org) (gcc version 7.3.1 20180303 (Red Hat 7.3.1-5) (GCC)) #1 SMP Tue Jun 5 16:30:48 UTC 2018
...
[  643.576470] chipsec: Unknown symbol valid_mmap_phys_addr_range (err 0)

Run:
insmod: ERROR: could not insert module /usr/lib64/python2.7/site-packages/chipsec-1.3.5-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module

BrentHoltsclaw, 2018-07-05:
Which version of chipsec are you running?  Can you please ensure that you have the following fixes within the driver https://github.com/chipsec/chipsec/commit/93fa9f19b581a53e67a1a3a036b6b855aecc1bf0

ismaws, 2018-07-05:
My mistake, I was pulling from latest stable branch (v1.3.6rc1). Let me try using the latest latest.

ismaws, 2018-07-05:
Confirmed - fix works. Closing.

thanks.

closed 2018-07-05 by ismaws

ismaws, 2018-09-21:
reopenning issue. 

I'm pulling from master branch (currently pointing to v1.3.6rc1) but I'm running kernel 4.1.12-124.19.4.el7uek.x86_64:

[    0.000000] Linux version 4.1.12-124.19.4.el7uek.x86_64 (mockbuild@) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-16.0.3) (GCC) ) #2 SMP Tue Sep 18 20:34:35 PDT 2018

4.1.12 does not get the fix which only applies to LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4)

build:

copying drivers/linux/include/chipsec.h -> /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
WARNING: could not find /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
WARNING: "valid_mmap_phys_addr_range" [/home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!

install:

copying drivers/linux/include/chipsec.h -> /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
WARNING: could not find /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
WARNING: "valid_mmap_phys_addr_range" [/home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!
copying /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko -> /home/user/chipsec/pkg/chipsec/build/lib.linux-x86_64-2.7/chipsec/helper/linux

run:
insmod: ERROR: could not insert module /usr/local/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module

[ 1010.003713] chipsec: Unknown symbol valid_mmap_phys_addr_range (err 0)

sudo insmod /usr/local/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko
insmod: ERROR: could not insert module /usr/local/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module

ErikBjorge, 2018-09-21:
I will have Brent look at this and do some testing.  Thanks for taking the time to reopen the issue.

BrentHoltsclaw, 2018-09-21:
I haven't had a chance to set up that version of of RHEL yet.  Could you please try the following?
change line 547 in chipsec_km.c
`#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4)
int valid_mmap_phys_addr_range(unsigned long pfn, size_t size)`
to 
`#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,12)
int valid_mmap_phys_addr_range(unsigned long pfn, size_t size)`
and rerun setup.py

afoodu2, 2019-01-02:
Hi, would it be possible to make this change to support 4.1.12 and 4.14 based kernels, which
suffer from this runtime issue:
chipsec: Unknown symbol valid_mmap_phys_addr_range (err 0)

--- drivers/linux/chipsec_km.c-o	2018-11-20 11:15:02.000000000 -0800
+++ drivers/linux/chipsec_km.c	2019-01-02 00:48:56.734436407 -0800
@@ -547 +547 @@
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4)
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,12) && defined(ARCH_HAS_VALID_PHYS_ADDR_RANGE)

BrentHoltsclaw, 2019-01-02:
Pull Request #499 should fix the issue once accepted.  Thanks for the help.