chipsec/chipsec #429: CHIPSEC port to MicroPython for execution in UEFI Shell

pietrushnic, 2018-08-12:
Since Tianocore published [initial MicroPython release](https://github.com/tianocore/tianocore.github.io/wiki/MicroPython-Test-Framework-for-UEFI) I would like to suggest CHIPSEC port to MicroPython for UEFI Shell validation environment.

It looks like edk2 is committed to using MicroPython as their scripting language of choice and they seem to build validation framework based on it. CHIPSEC can leverage that effort to seamlessly add and expand the test scope. Probably code base can be reduced since it doesn't make sense to fight with Python compatibility in edk2.

Unfortunately, there is no support for GCC yet, what IMO effectively limit the number of developers that can involve.

BrianRichardsonIntel, 2018-08-13:
We want to evaluate using MicroPython instead of CPython 2.7 for Python support under UEFI. The port supplied in the staging branch is the first step of that process. Any feedback you have on the current port is greatly appreciated. 

GCC support also needs to be resolved. I recommend filing an issue for this as a 'Feature Request' at bugzilla.tianocore.org. Please note in the request that this is for an item in edk2-staging (the eval branch of TianoCore github). I'm maintaining the docs for this project, and can add it to the 'to-do' list I'm creating in the wiki.

Thanks for your interest.

ErikBjorge, 2018-08-20:
We are starting work on supporting Python 3.4 as well as Python 2.7.  Once we have this done we will also enable OS helper support for MicroPython on the UEFI Shell.  This is going to be a long term project because we can't just dump Python 2.7 support at this time.

pietrushnic, 2018-08-23:
@BrianRichardsonIntel  I will fill issue on bugzilla as soon as I will get some time. I believe we can talk about Python and MicroPython stuff if you will find time on OSFC  2018, since I see you have talk there. I also plan to present a couple things one of it is [BITS and CHIPSEC as coreboot payloads](https://osfc.io/talks/bits-and-chipsec-as-coreboot-payloads). We already addressed BITS part and will move with CHIPSEC in the following week. I believe GRUB+MicroPython (+BIST/CHIPSEC) would be interested to industry and some customers of [3mdeb](https://3mdeb.com).

BrianRichardsonIntel, 2018-08-23:
Sounds good. Thanks.

Sae86, 2022-06-14:
Closing due to use of UEFI Port of Python 3.

closed 2022-06-14 by Sae86