chipsec/chipsec #435: PCI0.0.0.<REG_NAME> vs <REG_NAME>

gibarrasch, 2018-09-03:
Hi,
I was trying to figure out why some of the register looks like "PCI0.0.0.<REG_NAME>" instead of just <REG_NAME>.
Also I saw some potential issues related with this.
For instance the registers:
- PCI0.0.0_BDSM
- PCI0.0.0_BGSM
- PCI0.0.0_GGC
Which are defined in common.xml and used by the module MemConfig, seem to have overrides for the platforms: byt.xml, cht.xml, apl.xml but with the names BDSM, BGSM and GGC (just <REG_NAME>)

So, if this registers are the same thing, i think the override won't work.

ErikBjorge, 2018-09-04:
I think the exact reason for the register name prefix used in common may be lost to history.  I have a feeling they were trying to be exact in the register naming.

You are correct that the register names do need to match to allow the override to work correctly for the platform.  We will need to go and fix the register names in the platform configuration files or remove the additional platform definitions if they are not needed.  We may also want to try to normalize the register names so we are consistent.

ErikBjorge, 2018-12-21:
Closing issue.  This should be fixed in PR #473.  If the issues still exists please reopen the issue.

closed 2018-12-21 by ErikBjorge