chipsec/chipsec #439: chipsec_util uefi kernel-mode problem

mj-p, 2018-09-21:
Hey, when using chipsec_util I cannot use uefi and nvram in kernel-mode.
Other commands like cpu info do work so does chipsec_main!
Version: 1.3.6.rc1 (master)
System: Windows 10 Enterprise 1803
Platform: SNB

```
>python chipsec_util.py uefi var-list

[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
ERROR: chipsec kernel driver is not loaded (in native API mode?)
ERROR: chipsec kernel driver is not loaded (in native API mode?)
[CHIPSEC] Executing command 'uefi' with args ['var-list']
[CHIPSEC] Enumerating all EFI variables via OS specific EFI Variable API..
[CHIPSEC] Decoding EFI Variables..
[CHIPSEC] Variables are in efi_variables.lst log and efi_variables.dir directory
[CHIPSEC] (uefi) time elapsed 0.141
```

ErikBjorge, 2018-09-21:
Under Linux we have a mixed usage of kernel mode driver usage and native mode (no driver).  You do still need to run as root or use sudo to get this to work.  We could force the driver to always load (my preferred solution) and then use the -n flag to enable native mode similar to chipsec_main.  If you would like to try this you can modify requires_driver in chipsec/utilcmd/uefi_cmd.py like the code below.

```
    def requires_driver(self):
        # No driver required when printing the util documentation
        if len(self.argv) < 3:
            return False
        return True
```

Let me know if this works for you and we can look at making changes to the UEFI Util command.

mj-p, 2018-09-24:
@ErikBjorge Your fix indeed does work!

Sadly I didn't find the variable I was looking for.
Do you have any idea how to modify PhysicalBootOrder-59d1c24f-50f1-401a-b101-f33e0daed443 from Windows 10?

ErikBjorge, 2018-09-24:
Did your attempt to modify the variable using var-read/var-write fail?  If so, what error message did the variable interface return?

Depending on the platform it may not be possible from the OS.  UEFI provides a number of methods that render UEFI variables read only from the OS.

mj-p, 2018-09-24:
I tried var-find and it didn't find it. From Windows I can't access the variable, I tried several APIs. Works fine from linux though, I can find it under efivars..

ErikBjorge, 2018-09-24:
Do any of the variable accesses work from within Windows with the change that I proposed?  Just want to verify before I flag this as a bug.

I am glad you can get things working under Linux.

mj-p, 2018-09-25:
Yes that did work

OQAChipset, 2018-09-25:
May I ask if the master branch version would work with Skylake processor? Before there was  https://github.com/otcshare/chipsec-prerelease for Skylake processor but this directory does not exist anymore. Please advice.

ErikBjorge, 2018-09-26:
That is a driver build issue and should be resolved on master.  A kernel change required us to update our driver code.

Are you running on a Skylake server or desktop platform?  The server configuration files are not available in the public version of CHIPSEC.

ErikBjorge, 2018-09-26:
You can add the following lines to drivers/linux/chipsec_ko.c to see if this solves the driver build issue with the previously working tree.  The code should drop in at line 546 and rebuild the driver.

DIff below.

```
diff --git a/drivers/linux/chipsec_km.c b/drivers/linux/chipsec_km.c
index d99e053..6dd8d2b 100644
--- a/drivers/linux/chipsec_km.c
+++ b/drivers/linux/chipsec_km.c
@@ -544,6 +544,13 @@ int __weak phys_mem_access_prot_allowed(struct file *file,
         return 1;
 }
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4)
+int valid_mmap_phys_addr_range(unsigned long pfn, size_t size)
+{
+	return 1;
+}
+#endif
+
 #ifndef ARCH_HAS_VALID_PHYS_ADDR_RANGE
 static inline int valid_phys_addr_range(phys_addr_t addr, size_t count)
 {
```

OQAChipset, 2018-09-27:
I still got the same for the old combined release that I downloaded 
before. Did you mean I put these lines on that old combined release, or 
I should put them on the latest Master release?

  chipsec_main
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1-p0008
[CHIPSEC] Arguments:
insmod: ERROR: could not insert module 
/usr/lib64/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: 
Unknown symbol in module
ERROR: Could not start the OS Helper, are you running as Admin/root?
            Message: "Command '['insmod', 
'/usr/lib64/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko', 
'', '']' returned non-zero exit status 1"

uname  -r

4.1.12-124.19.5.el7uek.x86_64

Grace

On 9/25/2018 5:56 PM, Erik Bjorge wrote:
> |#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4) +int 
> valid_mmap_phys_addr_range(unsigned long pfn, size_t size) +{ + return 
> 1; +} +#endif|

BrentHoltsclaw, 2018-09-27:
The fix referenced above was committed in April so it would not be within the release and you would have to add it in.  Alternatively, you can build from the master branch.
If this does not fix the problem, could you please post the output of building the driver?  This will help debug the issue.
`python setup.py build_ext -i`

OQAChipset, 2018-09-28:
python setup.py build_ext -i
running build_ext
copying build/lib.linux-x86_64-2.7/chipsec/helper/linux/cores.so -> chipsec/helper/linux
copying build/lib.linux-x86_64-2.7/chipsec_tools/efi_compressor.so -> chipsec_tools
building the linux driver
creating /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers
creating /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
creating /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/amd64/cpu.asm -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/Makefile -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
creating /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/README -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/dkms.conf -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/WARNING.txt -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/installOutput -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/AUTHORS -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
creating /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/i386/cpu.asm -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/chipsec_km.cOLD -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec_km.c -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/COPYING -> /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux
WARNING: could not find /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
WARNING: "valid_mmap_phys_addr_range" [/root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!
copying /root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/root/chipsec-1.3.6rc1-p0008/build/lib.linux-x86_64-2.7/drivers' (and everything under it)

OQAChipset, 2018-10-24:
I tried downloading chipsec-master.zip from master branch. Recompiled with "python setup.py build_ext -i"and then install "python setup.py install". However, still it is not successfully run. Please see all the building, installing and running logs below:

python ./setup.py build_ext -i
running build_ext
building 'chipsec.helper.linux.cores' extension
creating build
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/chipsec
creating build/temp.linux-x86_64-2.7/chipsec/helper
creating build/temp.linux-x86_64-2.7/chipsec/helper/linux
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python2.7 -c chipsec/helper/linux/cores.c -o build/temp.linux-x86_64-2.7/chipsec/helper/linux/cores.o
chipsec/helper/linux/cores.c:22:0: warning: "_GNU_SOURCE" redefined [enabled by default]
 #define _GNU_SOURCE
 ^
<command-line>:0:0: note: this is the location of the previous definition
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/chipsec
creating build/lib.linux-x86_64-2.7/chipsec/helper
creating build/lib.linux-x86_64-2.7/chipsec/helper/linux
gcc -pthread -shared -Wl,-z,relro build/temp.linux-x86_64-2.7/chipsec/helper/linux/cores.o -L/usr/lib64 -lpython2.7 -o build/lib.linux-x86_64-2.7/chipsec/helper/linux/cores.so
building 'chipsec_tools.efi_compressor' extension
creating build/temp.linux-x86_64-2.7/chipsec_tools
creating build/temp.linux-x86_64-2.7/chipsec_tools/compression
creating build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA
creating build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/SDK
creating build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/SDK/C
creating build/temp.linux-x86_64-2.7/chipsec_tools/compression/Tiano
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_tools/compression/EfiCompressor.c -o build/temp.linux-x86_64-2.7/chipsec_tools/compression/EfiCompressor.o
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_tools/compression/LZMA/LzmaCompress.c -o build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/LzmaCompress.o
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_tools/compression/LZMA/LzmaDecompress.c -o build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/LzmaDecompress.o
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/Bra86.c -o build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/SDK/C/Bra86.o
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/LzFind.c -o build/temp.linux-x86_64-2.7/chipsec_tools/compression/LZMA/SDK/C/LzFind.o
gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Ichipsec_tools/compression/Include -I/usr/include/python2.7 -c chipsec_to [... cut ...]
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] Arguments:
insmod: ERROR: could not insert module /usr/lib64/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module
ERROR: Could not start the OS Helper, are you running as Admin/root?
           Message: "Command '['insmod', '/usr/lib64/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko', '', '']' returned non-zero exit status 1"

BrentHoltsclaw, 2018-10-25:
Thanks for the verbose information.  Were you able to patch the code that @ErikBjorge mentioned above?  Which distribution are you running? I'll try to reproduce the issue within a VM if you can give specifics.

BrentHoltsclaw, 2019-01-04:
It appears that both issues have been resolved.  Because the issues were crossed, please open a new issue if more assistance is needed.

closed 2019-01-04 by BrentHoltsclaw