chipsec/chipsec #445: TypeError on uefi decode

PerezFederico, 2018-10-01:
[68.zip](https://github.com/chipsec/chipsec/files/2435039/68.zip)

When running:

    chipsec_util uefi decode 68

I'm getting:

``` 
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Executing command 'uefi' with args ['decode', '68']

[CHIPSEC] Parsing EFI volumes from '68'..
Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.3.6rc1', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec_util.py", line 264, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec_util.py", line 231, in main
    comm.run()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 236, in run
    decode_uefi_region(_uefi, cur_dir, filename, fwtype)
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 539, in decode_uefi_region
    fwtype = identify_EFI_NVRAM( region_data )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 245, in identify_EFI_NVRAM
    (offset, size, hdr) = uefi_platform.EFI_VAR_DICT[ fw_type ]['func_getnvstore']( b )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi_platform.py", line 374, in getNVstore_NVAR
    cur_offset, next_offset, Name, Type, Attributes, State, Checksum, Size, FileImage, HeaderSize, UD, fCalcSum = NextFwFile(FvImage, FvLength, next_offset, polarity)
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/hal/uefi_common.py", line 709, in NextFwFile
    next_offset = align(fof + fsize, 8)
TypeError: unsupported operand type(s) for +: 'int' and 'tuple'
```

I tried with an older version and it seemes to be working. 

Did some research (a little, no testing) and I think that [this](https://github.com/chipsec/chipsec/commit/8fa75dec5c9108e69b43e1404c856e990757f21f#diff-d99006731e4e7b8e42dc3677d650581c) commit may be the issue.

Line 697: It seems that it unpacks a tuple and then does not select any element.

If someone confirms that this makes sense I can create the patch.

Thanks

BrentHoltsclaw, 2018-10-01:
Thanks.  
chipsec/hal/uefi_common.py line 703 appears to be incorrect:
`fsize = struct.unpack("Q",FvImage[fof+file_header_size:fof+file_header_size+struct.calcsize("Q")])`
should be
`fsize = struct.unpack("Q",FvImage[fof+file_header_size:fof+file_header_size+struct.calcsize("Q")])[0]`

PerezFederico, 2018-10-01:
Thanks. Creating PR

PerezFederico, 2018-10-02:
Status update: 
Did not created PR yet because when you apply this change, the program loops forever, at least with the test file I provided.

It seems that the size returned by the unpack is wrong. 
I will keep posting what I find.

Thanks

ErikBjorge, 2018-10-02:
Can you provide the test file so we can also look at the issue?

ErikBjorge, 2018-10-02:
Sorry, missed the fact that the file was included at the top of the issue.

BrentHoltsclaw, 2018-10-05:
It seems that there is invalid information within the FV headers of the file supplied.  Have you been able to get any other tools to properly open the file as I am having trouble (UEFITool, VolInfo, uefi-firmware-parser)?  I'm looking into adding some logic to call out when the header information is invalid and properly exit.

PerezFederico, 2018-10-05:
Tried now, and it seems to fail with uefi-firmware-parser.
It is possible that it is a faulty image because it is extracted with binwalk from another binary.

I think that it would be nice to properly exit as you mention, but also dump whatever is possible before the error is detected.

BrentHoltsclaw, 2018-10-06:
@PerezFederico can you please check out PR#451 and see if that helps

0xfede7c8, 2018-11-22:
Fixed on #451.
Closing.

closed 2018-11-22 by 0xfede7c8