chipsec/chipsec #453: compilation errors.  [1]: PyBufferProcs’ {aka ‘struct <anonymous>’} has no member named ‘bf_getsegcount’

jamespharvey20, 2018-10-11:
Cloned git master 710b115.  Arch linux.  linux 4.18.12.arch1-1.  gcc 8.2.1+20180831-1.  

`python setup.py build_ext -i` fails, with output: http://termbin.com/nxfj

I believe this is supposed to be with python3, but just in case, `python2 setup.py build_ext -i` fails, with output: http://termbin.com/qgx6

jamespharvey20, 2018-10-11:
Thanks, sorry about that.  Building from Arch's AUR doesn't seem to handle the dependencies properly.  I'll look into that.  Building manually with the dependencies got past this.

I am getting a warning in the build now though.  Full output here: http://termbin.com/n4ox - pertinent section below

Not sure if the warning really matters though.  `sudo python2 chipsec_main.py` runs with output http://termbin.com/39d7 -- not sure yet if anything is missing due to the warning given.

```
building the linux driver
creating /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers
creating /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/AUTHORS -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/COPYING -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/Makefile -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/README -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/WARNING.txt -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/amd64/cpu.asm -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/chipsec_km.c -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/dkms.conf -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/i386/cpu.asm -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/i386
creating /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/include
WARNING: could not find /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
copying /home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/home/jamespharvey20/chipsecb.git/build/lib.linux-x86_64-2.7/drivers' (and everything under it)
```

jamespharvey20, 2018-10-11:
In the output is a few `not supported in this platform` messages.  Dual Intel Xeon CPU E5-2690 v1 (2.9GHz) CPUs.  ASRock EP2C602 motherboard, with Intel C602 chipset.  Don't know if this is support that could be added to chipset or not.

```
...
[*] running module: chipsec.modules.common.rtclock
Skipping module chipsec.modules.common.rtclock since it is not supported in this platform
...
[*] running module: chipsec.modules.common.smm
Skipping module chipsec.modules.common.smm since it is not supported in this platform
...
[*] running module: chipsec.modules.debugenabled
[*] NOT IMPLEMENTED: CPU Debug features are not supported on this platform
Skipping module chipsec.modules.debugenabled since it is not supported in this platform

[*] running module: chipsec.modules.memconfig
Skipping module chipsec.modules.memconfig since it is not supported in this platform

[*] running module: chipsec.modules.remap
Skipping module chipsec.modules.remap since it is not supported in this platform

[*] running module: chipsec.modules.smm_dma
Skipping module chipsec.modules.smm_dma since it is not supported in this platform
...
[CHIPSEC] Modules not implemented 7:
[*] NOT IMPLEMENTED: chipsec.modules.common.rtclock
[*] NOT IMPLEMENTED: chipsec.modules.common.smm
[*] NOT IMPLEMENTED: chipsec.modules.common.secureboot.variables
[*] NOT IMPLEMENTED: chipsec.modules.debugenabled
[*] NOT IMPLEMENTED: chipsec.modules.memconfig
[*] NOT IMPLEMENTED: chipsec.modules.remap
[*] NOT IMPLEMENTED: chipsec.modules.smm_dma
...
```

`chipsec.modules.common.secureboot.variables` is just because none of the variables exist because secureboot isn't enabled yet.

ErikBjorge, 2018-10-11:
The warning that you get is expected and does not impact the functionality of the driver.  We may need to update our build to check if the build option is supported.

We can look at adding support for Xeon processors in some of these tests or create Xeon specific versions.  I many cases the registers we need to test a Xeon processor differ from the registers used for Core processors.  We just need to enhance the test to support either register layout.

As for the secure boot test, the warning is because the variables do not exist.

ErikBjorge, 2018-12-21:
I would like to file a new issue for updating some of the existing tests to support Xeon platforms and close this issue.  Does this seem reasonable?

jamespharvey20, 2018-12-21:
Sounds good to me!

Sae86, 2022-05-23:
Closing as this issue seems to have been resolved. I am also opening #1480

closed 2022-05-23 by Sae86