chipsec/chipsec #456: CFL unsupported

GHswitt, 2018-10-12:
Hi,

it seems CFL is not supported or not tested:

1. FREG base/limit size was not 15 bits -> wrong BIOS base address/limit (see pull request)
2. Can't read ABASE and SBREGBAR -> The PCI devices are hidden and it isn't possible to make the visible again (locked):
[iobar] read ABASE + 0x30 (4)
[pci] reading B/D/F: 0/31/2, offset: 0x40, value: 0xFFFFFFFF
[iobar] ABASE: 0xFF00 (size = 0x100)
[io] IN 0xFF30: value = 0xFFFFFFFF, size = 0x04
...
[pci] reading B/D/F: 0/31/1, offset: 0x00, value: 0xFFFFFFFF
[pci] reading B/D/F: 0/31/1, offset: 0xE1, value: 0xFF
[pci] writing B/D/F: 0/31/1, offset: 0xE1, value: 0xFE
[pci] reading B/D/F: 0/31/1, offset: 0x10, value: 0xFFFFFFFF
[mmio] SBREGBAR: 0x00000000FF000000 (size = 0x1000000)

Any idea how to solve this? I'm using a fixed value for now...

GHswitt, 2018-10-12:
#455 and #457

ErikBjorge, 2018-10-12:
When we tested originally the ability to un-hide the P2SB device was functional.  If you use the mm_msgbus option from msgbus in chipsec_util do you see the same issue.  I just want to see how the un-hide code is failing.  You should be able to run the command below and see what happens without your patches.

python chipsec_util.py -v msgbus mm_read 0xBC 0x0980

If this is failing the only other option is to use a fixed address.

GHswitt, 2018-10-15:
Here is the result:

[CHIPSEC] MMIO msgbus read: port 0xBC + 0x00000980
[pci] reading B/D/F: 0/31/1, offset: 0x00, value: 0xFFFFFFFF
[pci] reading B/D/F: 0/31/1, offset: 0xE1, value: 0xFF
[pci] writing B/D/F: 0/31/1, offset: 0xE1, value: 0xFE
[pci] reading B/D/F: 0/31/1, offset: 0x10, value: 0xFFFFFFFF
[mmio] SBREGBAR: 0x00000000FF000000 (size = 0x1000000)
[mmio] 0xFF000000 + 0x00BC0980 = 0xFFFFFFFF
[pci] reading B/D/F: 0/31/1, offset: 0x00, value: 0xFFFFFFFF
[pci] reading B/D/F: 0/31/1, offset: 0xE1, value: 0xFF
[pci] writing B/D/F: 0/31/1, offset: 0xE1, value: 0xFF
[CHIPSEC] result: 0xFFFFFFFF
[CHIPSEC] (msgbus) time elapsed 0.000
[helper] Linux Helper stopped/unloaded
[helper] Linux Helper deleted

I can only make it visible if I build the FW without locking the SB device. The RC FW does not use  0xE0 Bit 8 to hide the device. It seems the complete PCIe EP device configuration access is disabled and locked.

GHswitt, 2018-10-15:
Here is a log file (with modified cfl.xml). Still have a problem accessing the RTC config (0xFDC33400/0xFDC33414).

[20181015_debug.txt](https://github.com/chipsec/chipsec/files/2478044/20181015_debug.txt)

ErikBjorge, 2018-10-15:
Thanks for providing the additional information on the P2SB device configuration.  I now understand how the system is configured.  Sorry for not understanding this originally. Switching to a fixed address is the correct solution.  You still should leave the definitions for the P2SB registers in the configuration file.

I will do some testing on a CFL system here and check some differences in system configuration.

ErikBjorge, 2018-10-16:
I did some more testing on a CFL platform and can't successfully validate the changes to PWRMBASE and PMC_SHDW_BAR4.  All reads return 0xFFFFFFFF even though I verified that the correct base was now being used via the fixed_address value provided.  Other registers such as the GPIO registers seem to function correctly when accessing them via SBREGBAR.

Can you send me the platform information so I can see if we are testing with a different PCH?  I just need the last six lines of output.

python chipsec_util.py platform

ErikBjorge, 2018-10-23:
I am still unable to get the changes to CFL to function as expected.  I will continue to investigate.  Your other changes have been merged.

FTS-ML, 2019-02-21:
Hello @ErikBjorge 

I'm running into the same issues on a CoffeeLake platform - also using C246. Do you have any new ideas how to proceed?

Thanks & Regards,
Michael

ErikBjorge, 2019-03-07:
Thanks for the reminder on this and sorry for letting this slip for so long.

Sae86, 2022-06-02:
Closing as CLF support has been added.