chipsec/chipsec #461: AMD and/or ARM support?

ismaws, 2018-10-19:
This is not an issue, more of a future-release / other tools question:

Does chipsec have any plans to support non-intel architectures? 
Are there any other tools specific to check secure cfg or AMD/ARM architectures? also, are there other tools that complement chipsec in its current scope?

ErikBjorge, 2018-10-20:
We do have plans for adding the ability to test other architectures to CHIPSEC.  Part of this will require restructuring the driver and configuration file layout and updating the detection process.  We also need a simple method for tagging modules so they run only under the appropriate configurations.  I am sure we will find other issues as we start to add these features.

If you would like we can discuss potential changes in this issue for now.  At some point we may need to track plan/progress on the wiki or some other location.

JamesMihm, 2019-03-20:
I would like to use CHIPSEC for auditing arm based Baseboard Management Controllers (BMCs) configurations. Is there any update on the progress for adding arm support?

In addition to adding arm support, there needs to be support for the detecting or specifying which soc is being used; such as AST2x00, Pilot X, or the Poleg BMC.

ErikBjorge, 2019-03-21:
@BrentHoltsclaw and I were discussing this today and are working on a plan.  This has taken a lower priority due to workloads but has not been forgotten.  We have a few tasks to how the tool currently works so new architectures can be added.  Once that is done we will need help adding ARM specific drivers and tests.  Expect some of the layout changes in the next month or so.

Sae86, 2022-06-14:
AMD support added by @kerneis-anssi

ke88yun, 2024-11-08:
When will the support for ARM based PC be added to the framework?

npmitche, 2024-11-08:
Currently we would ask for community contributions or an ecosystem partner to add support.  If you need specific examples to for adding features please propose a meeting or join the community meeting!