chipsec/chipsec #465: Running tests without kernel module

JohnAZoidberg, 2018-10-24:
I would like to build the package without the kernel module.
For building without kernel modules there's the option `skip-driver`, so that works fine, but when I execute the tests it tries to build the module and the `skip-driver` option is not accepted for `python2 setup.py test --skip-driver`.
This causes the tests to fail if I don't have the kernel headers installed.

JohnAZoidberg, 2018-10-24:
Yeah sure, I can disable the tests for the time being in this use-case.

BrentHoltsclaw, 2018-10-30:
Can you try `python setup.py build --skip-driver test` and let me know if that is the functionality you are looking for?

JohnAZoidberg, 2018-10-30:
Yes, thank you, that works and seems to do what I want.

closed 2018-10-30 by JohnAZoidberg