chipsec/chipsec #479: chipec_util uefi decode crash

0xfede7c8, 2018-11-22:
With the example provided below I'm getting:

``` bash
➜  sudo chipsec_util uefi decode IMAGE9K.ROM
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'uefi' with args ['decode', 'IMAGE9K.ROM']

[CHIPSEC] Parsing EFI volumes from 'IMAGE9K.ROM'..
Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.3.6', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec_util.py", line 261, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec_util.py", line 228, in main
    comm.run()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 243, in run
    decode_uefi_region(_uefi, cur_dir, filename, fwtype)
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 533, in decode_uefi_region
    parse_uefi_region_from_file( _uefi, fname, fwtype, fv_pth )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 515, in parse_uefi_region_from_file
    tree = build_efi_model(_uefi, rom, fwtype)
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 400, in build_efi_model
    model = build_efi_tree( _uefi, data, fwtype )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 344, in build_efi_tree
    fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 315, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 266, in build_efi_modules_tree
    guid0, guid1, guid2, guid3, sec.DataOffset, sec.Attributes = struct.unpack(EFI_GUID_DEFINED_SECTION, sec.Image[sec.HeaderSize:sec.HeaderSize+EFI_GUID_DEFINED_SECTION_size])
struct.error: unpack requires a string argument of length 20
```

Target file: 
[IMAGE9K.zip](https://github.com/chipsec/chipsec/files/2609194/IMAGE9K.zip)

ErikBjorge, 2018-11-26:
Thanks for providing the failing image.  We will take a look at what is going on.

BrentHoltsclaw, 2018-11-26:
For a temp fix try changing line 266 from
`                guid0, guid1, guid2, guid3, sec.DataOffset, sec.Attributes = struct.unpack(EFI_GUID_DEFINED_SECTION, sec.Image[sec.HeaderSize:sec.HeaderSize+EFI_GUID_DEFINED_SECTION_size])`
to
`                guid0, guid1, guid2, guid3 = struct.unpack("<IHH8s", sec.Image[sec.HeaderSize:sec.HeaderSize+16])`

I need to do some more testing to make sure it doesn't break any other functionality.

0xfede7c8, 2018-11-26:
Are you sure? What happens with `sec.DataOffset` and `sec.Attributes`? It will fail because their are used afterwads.

BrentHoltsclaw, 2018-11-26:
In the case of the file provided, it works.  I need to work on a better long term fix.

0xfede7c8, 2018-11-27:
Ok. Will implement a try/catch with the provided version as a fallback so it can handle both cases.

0xfede7c8, 2018-11-27:
If this is a good enough solution I can contribute it.

BrentHoltsclaw, 2018-11-27:
please see PR 483

closed 2018-11-28 by 0xfede7c8