chipsec/chipsec #480: FV header doesn't validate the checksum

gibarrasch, 2018-11-26:
Hi Guys, I noticed that the header validation of the FV is disabled. is there any reason?

https://github.com/chipsec/chipsec/blob/master/chipsec/hal/uefi_common.py#L579

ErikBjorge, 2018-11-26:
I agree that we should validate the checksum and produce a warning if it is bad.

0xfede7c8, 2018-11-27:
I have another question about this. Most of the times the check is disabled by the header Attributes.
Only if the attributes have the FFS_ATTRIB_CHECKSUM flag, the checksum is calculated and compared. If not, it is hardcoded to FFS_FIXED_CHECKSUM.

Why does this happens?

skochinsky, 2018-11-27:
This is a question to UEFI spec writers. At a guess, checksum calculation over the whole file can be  expensive so it was made optional and is used sparingly. Besides, the 8-bit sum of bytes is not very robust and may fail in very simple  corruption cases. Some vendors use a CRC32 wrapper section when they need a reliable integrity check of the file's data.

0xfede7c8, 2018-11-28:
Great thanks!

closed 2026-10-12 by dscott90