chipsec/chipsec #481: Physical memory write doesn't work on Windows

hardboost, 2018-11-26:
1. allocate phys mem
2. write **0xaabbccdd** into it
3. read returns **0x19**

```
C:\Users\user\Desktop\chipsec>chipsec_util mem allocate 0x1000

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'mem' with args ['allocate', '0x1000']

[CHIPSEC] Allocated 1000 bytes of physical memory: VA = 0xFFFFC480E4835000, PA = 0x000000017EF53000
[CHIPSEC] (mem) time elapsed 0.000

C:\Users\user\Desktop\chipsec>chipsec_util mem writeval 0x000000017EF53000 4 0xaabbccdd

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'mem' with args ['writeval', '0x000000017EF53000', '4', '0xaabbccdd']

[CHIPSEC] writing 4-byte value 0xAABBCCDD to PA 0x000000017EF53000..
[CHIPSEC] (mem) time elapsed 0.000

C:\Users\user\Desktop\chipsec>chipsec_util mem readval 0x000000017EF53000 4

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'mem' with args ['readval', '0x000000017EF53000', '4']

[CHIPSEC] reading 4-byte value from PA 0x000000017EF53000..
[CHIPSEC] value = 0x19
[CHIPSEC] (mem) time elapsed 0.000
```

From ipython:
```
In [6]: from chipsec import chipset
In [7]: cs = chipset.cs()
In [8]: cs.init(None, None, True)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API

In [9]: va, pa = cs.helper.alloc_physical_mem(0x1000, 0xffffffff)
In [10]: cs.helper.write_physical_mem(pa, 4, '\xaa\xbb\xcc\xdd')
Out[10]: ''
In [11]: cs.helper.read_physical_mem(pa, 4).encode('hex')
Out[11]: '18000000'
```

On CHIPSEC 1.3.6rc1 everything worked as expected. Driver for 1.3.6rc1 was compiled with WDK, and driver for 1.3.6 was compiled with VS2017.

ErikBjorge, 2018-11-26:
Can you please try building the driver using the WDK to see if it is a tool chain issue?

hardboost, 2018-11-27:
> Can you please try building the driver using the WDK to see if it is a tool chain issue?

Tried Free and Checked builds: everything the same. I have some debugging information:
```
In [18]: va, pa = cs.helper.alloc_physical_mem(0x1000, 0xffffffff)

In [19]: pa
Out[19]: 2023038976 (0x78952000)

In [20]: mem_dump(pa, 0x20)
00000000: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000010: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................

In [21]: cs.helper.write_physical_mem(pa, 0x20, '\xaa' * 0x20)

In [22]: mem_dump(pa, 0x20)
00000000: 18 00 00 00 00 20 95 78  20 00 00 00 AA AA AA AA  ..... .x .......
00000010: AA AA AA AA AA AA AA AA  AA AA AA AA AA AA AA AA  ................
```

As you can see, it writes the following data:
1. 0x00000018 - ?
2. 0x78952000 - physical address of destination
3. 0x00000020 - size of data
4. data itself

hardboost, 2018-11-27:
I guess the mistake is here:
https://github.com/chipsec/chipsec/blob/4761a38aa2414ffc0ebb22a40b8a25818b167845/drivers/win7/driver.c#L486
It will write `HighPart + 0x18` into `HighPart`. So, it will be 0x18 in case PhysicalAddress is below 4GB, and 0x19 if it is above 4GB range. And `pInBuf` pointer is not changed.
Probably, it should be like the following:
```c
pInBuf = pInBuf + (3 * sizeof(UINT32));
```
At least it works well for me.

BrentHoltsclaw, 2018-11-27:
Thanks for looking into the issue.  I created a pull request to get it fixed.

ErikBjorge, 2018-12-21:
Fixed in PR #482.  Please reopen if this is still an issue after updating to the latest code.

closed 2018-12-21 by ErikBjorge