chipsec/chipsec #493: chipsec smi fuzz has found bad SMI for every SMI handler.

drmint80, 2018-12-18:
When I was test SMI fuzzer that has found potential occurrences of unchecked input pointers for any SMI handler. 
example:
Fuzzing for 0x20 SMI handler
chipsec_main -m tools.smm.smm_ptr -a fuzz,0x20:0x40 -l lognew.txt

[*] fuzzing SMI# 0x20 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x00000000DA095000
    > SMI 20 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x00000000DA095000
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A
    < checking buffers
    contents changed at 0x00000000DA095000 +[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]
    restoring 0x500 bytes at 0x00000000DA095000
[!] DETECTED: SMI# 20 data 0 (rax=5A5A5A5A5A5A5A5A rbx=DA095000 rcx=0 rdx=5A5A5A5A5A5A5A5A rsi=5A5A5A5A5A5A5A5A rdi=5A5A5A5A5A5A5A5A)
[!] Potentially bad SMI detected! Stopped fuzing (see FUZZ_BAIL_ON_1ST_DETECT option)
[-] <<< Done: found 1 potential occurrences of unchecked input pointers

then 
fuzz 0x4 smi handler
chipsec_main -m tools.smm.smm_ptr -a fuzz,0x4:0x40 -l lognew.txt

[*] fuzzing SMI# 0x04 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x00000000DA093000
    > SMI 04 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x00000000DA093000
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A
    < checking buffers
    contents changed at 0x00000000DA093000 +[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]
    restoring 0x500 bytes at 0x00000000DA093000
[!] DETECTED: SMI# 4 data 0 (rax=5A5A5A5A5A5A5A5A rbx=DA093000 rcx=0 rdx=5A5A5A5A5A5A5A5A rsi=5A5A5A5A5A5A5A5A rdi=5A5A5A5A5A5A5A5A)
[!] Potentially bad SMI detected! Stopped fuzing (see FUZZ_BAIL_ON_1ST_DETECT option)
[-] <<< Done: found 1 potential occurrences of unchecked input pointers

also fuzz 0x35 smi handler
chipsec_main -m tools.smm.smm_ptr -a fuzz,0x35:0x40 -l lognew.txt

[*] fuzzing SMI# 0x35 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x00000000DA091000
    > SMI 35 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x00000000DA091000
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A
    < checking buffers
    contents changed at 0x00000000DA091000 +[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]
    restoring 0x500 bytes at 0x00000000DA091000
[!] DETECTED: SMI# 35 data 0 (rax=5A5A5A5A5A5A5A5A rbx=DA091000 rcx=0 rdx=5A5A5A5A5A5A5A5A rsi=5A5A5A5A5A5A5A5A rdi=5A5A5A5A5A5A5A5A)
[!] Potentially bad SMI detected! Stopped fuzing (see FUZZ_BAIL_ON_1ST_DETECT option)
[-] <<< Done: found 1 potential occurrences of unchecked input pointers

ErikBjorge, 2018-12-18:
What OS and version of CHIPSEC are you using to perform your testing?  Also, what type of BIOS or system are you using?

ErikBjorge, 2018-12-20:
Thanks for the information.  We will look at this and see if we can determine what is happening.  I am hoping this is just a simple bug in our Windows driver code.

hardboost, 2018-12-21:
i'm sure this is related to https://github.com/chipsec/chipsec/pull/482 (it is not in release yet)

ErikBjorge, 2018-12-21:
Can you pull the latest code and try rebuilding with the change from #482 and see if it resolves the failure?  This would let us know if we have more issues to resolve.

BrentHoltsclaw, 2019-10-10:
This issue should be fixed within the latest code.  Please reopen if the problem has not been resolved.

closed 2019-10-10 by BrentHoltsclaw