chipsec/chipsec #498: 4.1.12 and 4.14 based kernels failing with Unknown symbol valid_mmap_phys_addr_range

afoodu2, 2019-01-02:
With v1.3.6rc1, chipsec.ko fails to load in 4.1.12 and 4.14 based kernels
due to undefined symbol valid_mmap_phys_addr_range.
Here's failing test output:

[root@sca05-0a81fac2 chipsec-1.3.6-p0010]# chipsec_main -i -m chipsec.modules.common.cpu.spectre_v2
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6-p0010
[CHIPSEC] Arguments: -i -m chipsec.modules.common.cpu.spectre_v2
insmod: ERROR: could not insert module /usr/lib64/python2.7/site-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module
ERROR: Could not start the OS Helper, are you running as Admin/root?
           Message: "Command '['insmod', '/usr/lib64/python2.7/site-packages/chipsec-1.3.6-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko', '', '']' returned non-zero exit status 1"
[root@sca05-0a81fac2 chipsec-1.3.6-p0010]# dmesg|tail -1
[40908.542086] chipsec: Unknown symbol valid_mmap_phys_addr_range (err 0)
[root@sca05-0a81fac2 chipsec-1.3.6-p0010]# 

With this code change:

--- drivers/linux/chipsec_km.c-o	2018-11-20 11:15:02.000000000 -0800
+++ drivers/linux/chipsec_km.c	2019-01-02 00:48:56.734436407 -0800
@@ -547 +547 @@
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,15,4)
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,12) && defined(ARCH_HAS_VALID_PHYS_ADDR_RANGE)

Here's successful test output:

[root@sca05-0a81fac2 chipsec-1.3.6-p0010]# chipsec_main -i -m chipsec.modules.common.cpu.spectre_v2
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6-p0010
[CHIPSEC] Arguments: -i -m chipsec.modules.common.cpu.spectre_v2
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.1.12-124.18.3.el7uek.x86_64 #2 SMP Sat Aug 11 08:03:54 PDT 2018 x86_64
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v5 v6 (Skylake / Cascade Lake CPU)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2020
[CHIPSEC] PCH     : Intel C621 (C620 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A1C1
 
[+] loaded chipsec.modules.common.cpu.spectre_v2
[*] running loaded modules ..

[*] running module: chipsec.modules.common.cpu.spectre_v2
[x][ =======================================================================
[x][ Module: Checks for Branch Target Injection / Spectre v2 (CVE-2017-5715)
[x][ =======================================================================
[*] CPUID.7H:EDX[26] = 1 Indirect Branch Restricted Speculation (IBRS) & Predictor Barrier (IBPB)
[*] CPUID.7H:EDX[27] = 1 Single Thread Indirect Branch Predictors (STIBP)
[*] CPUID.7H:EDX[29] = 0 IA32_ARCH_CAPABILITIES
[+] CPU supports IBRS and IBPB
[+] CPU supports STIBP
[-] CPU doesn't support enhanced IBRS
[!] WARNING: CPU supports mitigation (IBRS) but doesn't support enhanced IBRS
[!] OS may be using software based mitigation (eg. retpoline)

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.007
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   0:
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   1:
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] *****************************************************************
[root@sca05-0a81fac2 chipsec-1.3.6-p0010]#

BrentHoltsclaw, 2019-01-02:
Would you like to create a PR with the change or would you like us to create one?

afoodu2, 2019-01-02:
Please create one, as I'm not familiar with the process.
Thanks,