chipsec/chipsec #505: CHIPSEC must not assume PCI bus numbers of devices on server platforms

ErikBjorge, 2019-01-07:
Server platforms may place PCI devices on different buses than what is assumed by CHIPSEC.  Examples include the bus number of the PCH or PCI devices on processors other than socket 0.

CHIPSEC should update the PCI HAL to determine the correct bus for these devices.  @elicn suggests reading MSR 0x300 to resolve this issue.

Thanks @elicn for pointing this issue out.

frinzell, 2019-03-04:
Some routines are written that I need to get reviewed and tested.  Testing will take the longest since there are proposed changes to some of the core routines.

frinzell, 2019-05-17:
PR #566 submitted and open for review.

BrentHoltsclaw, 2019-10-18:
Chipsec should now support this feature.  Closing issue.   Please reopen if problem exists.

closed 2019-10-18 by BrentHoltsclaw