chipsec/chipsec #520: Missing 100 series PCH DID

gibarrasch, 2019-01-30:
I have a machine with PCH DID equal to 0xA152, I checked the documentations (from 2016), but It seems it is too old and it doesn't have references to this DID. Do you have newer documentation or contact with Intel to validate it?

ErikBjorge, 2019-01-30:
Thanks for providing the missing device ID.  I will dig through the documentation and see what I can find.  If I can find it I will submit a pull request to add it.

skochinsky, 2019-01-31:
https://www.intel.com/content/www/us/en/chipsets/100-series-chipset-datasheet-vol-1.html from Dec 2018 mentions A141-A51F for "LPC or eSPI controller" on PCH-H but lists no specific variant for A152 (last one is A150 for CM236).

However, after downloading [latest INF files](https://downloadcenter.intel.com/download/28182/Chipset-INF-Utility?product=1145) I found a few more in `SunrisePoint-HSystem.inf`:

> Intel(R) 100 Series/C230 Series Chipset Family LPC Controller (QMS180) - A151
> Intel(R) 100 Series/C230 Series Chipset Family LPC Controller (HM175) - A152
> Intel(R) 100 Series/C230 Series Chipset Family LPC Controller (QM175) - A153 
> Intel(R) 100 Series/C230 Series Chipset Family LPC Controller (CM238) - A154 
> Intel(R) 100 Series/C230 Series Chipset Family LPC Controller (QMU185) - A155

BrentHoltsclaw, 2019-12-06:
DID were added.  Closing issue.

closed 2019-12-06 by BrentHoltsclaw