chipsec/chipsec #523: Chipsec SMM modules working only on config options in every other arguments

bikashdash, 2019-02-01:
Hey guys,
 I am working on chipsec from quite sometime. recently i updated the latest release . 
It seems that, for every command line arguments, it i s only going on config options. if you see below i have choosen fuzz option in the command line. but evevn though it is hitting on config options

Anyone having same issue here?

C:\Users\Cr4sh\Documents\CPG_Work\chipsec>python chipsec_main.py -m tools.smm.smm_ptr -a fuzz,0:0x40
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6
[CHIPSEC] Arguments: -m tools.smm.smm_ptr -a fuzz,0:0x40

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Windows 10 10.0.17763 AMD64
[CHIPSEC] Platform: Mobile 7th Generation Core Processor (Kabylake H)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5910
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A154

[+] loaded chipsec.modules.tools.smm.smm_ptr
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.smm.smm_ptr
[*] Module arguments (1):
['fuzz,0:0x40']
[x][ =======================================================================
[x][ Module: A tool to test SMI handlers for pointer validation vulnerabilies
[x][ =======================================================================
Usage: chipsec_main -m tools.smm.smm_ptr [ -a <mode>,<config_file>|<smic_start:smic_end>,<size>,<address> ]
  mode          SMI handlers testing mode
    = config    use SMI configuration file <config_file>
    = fuzz      fuzz all SMI handlers with code in the range <smic_start:smic_end>
    = fuzzmore  fuzz mode + pass '2nd-order' pointers within buffer to SMI handlers
  size          size of the memory buffer (in Hex)
  address       physical address of memory buffer to pass in GP regs to SMI handlers (in Hex)
    = smram     pass address of SMRAM base (system may hang in this mode!)

hello fuzz,0:0x40
[*] Allocated memory buffer (to pass to SMI handlers)       : 0x0000000077390000
[*] Allocated 2nd buffer (address will be in the 1st buffer): 0x0000000077388000

[*] Configuration
    SMI testing mode          : config
    Config file           : chipsec/modules/tools/smm/smm_config.ini
    Memory buffer pointer     : 0x0000000077390000 (address passed in GP regs to SMI)
    Filling/checking memory?  : YES
      Second buffer pointer   : 0x0000000077388000 (address written to memory buffer)
      Number of bytes to fill : 0x500
      Byte to fill with       : 0x11
    Additional options (can be changed in the source code):
      Fuzzing SMI functions in ECX?          : 1
      Max value of SMI function in ECX       : 0x10
      Max value of SMI data (B3)             : 0x100
      Max offset of the pointer in the buffer: 0x20
      Passing pointer in all GP registers?   : 0
      Default values of the registers        : 0x5A5A5A5A5A5A5A5A
      Dump all register values every SMI     : 1
      Bail on first detection                : 1

[*] >>> Testing SMI handlers defined in 'chipsec/modules/tools/smm/smm_config.ini'..

[*] testing SMI# 0x00 (data: 0x00) SMI_name (custom_description)
[*] writing 0x500 bytes at 0x0000000077390000 -> PTR at +0x10 -> SIG at +0x0
[*] writing buffer at PA 0x0000000077388000 with 0x500 bytes ''
    > SMI 00 (data: 00)
      RAX: 0x0000000077390000
      RBX: 0x0000000077390000
      RCX: 0x0000000077390000
      RDX: 0x0000000077390000
      RSI: 0x0000000077390000
      RDI: 0x0000000077390000
    < checking buffers
[+] <<< Done: didn't find unchecked input pointers in tested SMI handlers

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.162
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   0:
[CHIPSEC] Modules passed          1:
[+] PASSED: chipsec.modules.tools.smm.smm_ptr
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] *****************************************************************

bikashdash, 2019-02-01:
After reviewing the code , i got to know that if we write the commands without "," symbols, it worked.
python chipsec_main.py -m tools.smm.smm_ptr -a fuzz 0x20:0x40

but the code says, comma should be there which is creating confusion. Moderators, please give your inputs.

chipsecintel, 2019-02-01:
It looks like we broke the option parsing support when we switched to using argparse.  We will work on restoring the legacy functionality.

Can you please file an issue on GitHub so we can track this?  If you don’t feel like filing the issue I can take care of it today.

Thanks,
-Erik

bikashdash, 2019-02-01:
Thanks for the input. I will try to file an issue as soon as I will be free
from the work..

ErikBjorge, 2019-02-01:
Sorry, I did not read the subject clearly this morning when I saw the mail.  This issue that you filed is just fine for us to track the problem.  Sorry for any confusion.

bikashdash, 2019-02-01:
Okay got it.

bikashdash, 2019-02-10:
Thanks folks for resolving issues

closed 2019-02-10 by bikashdash