chipsec/chipsec #527: SBREG BAR value does not take hidden P2SB into account

elicn, 2019-02-07:
CHIPSEC `mmio` module reads `SBREG_BAR` value from the P2SB device config space without un-hiding it first.

`SBREG_BAR` value is read from the P2SB config space, but most of the times P2SB is hidden by BIOS so it returns `FF`s for all reads. Since SBREG BAR value is 16 MiB aligned, the returned `FF`s are masked and shown as FF00_0000h. After un-hiding P2SB the value changes to the correct one.

To fix that, CHIPSEC would need to check whether P2SB is hidden (i.e. both VID and DID return `FF`s) and un-hide it before reading the value(s), and of course re-hiding it when necessary.

ErikBjorge, 2019-02-07:
What platform are you running this on?  We currently have support for one of the methods of hiding/unhiding the P2SB device that just requires an additional register definition to be added to the configuration file.  Other methods also exist for hiding devices in the PCH that we currently do not support.

Are you referring to the Hide bit in the P2SB function (Offset 0xE0 bit 8) or the function hide register for all the PCH devices?

elicn, 2019-02-07:
Running on a SKX platform, which is treated by CHIPSEC as a SKL for that matter.
I am referring to the hide bit you mentioned, which is actually [and sorry for being that petty] offset E1h bit 0 *. After un-hiding P2SB manually, the right value shows up on the mmio list.

* this is the difference between accessing the E0h register as a DWORD and overwriting the whole 32 bits, as opposed to accessing E1h as a BYTE and overwriting only 8 of them - which is actually the method Intel spec recommends.

ErikBjorge, 2019-02-11:
Sorry for the delay in getting back to you @elicn.

You are correct that accessing offset 0xE1 bit 0 is the preferred way of accessing the hide bit.  We did previously access the entire 32 bit register to hide/show the device but we corrected this.  We now use a special register definition named P2SB_HIDE that should be the single byte register definition.  This is accessed in the __hide_p2sb function in msgbus.py.  It also requires that the P2SBC register is also defined.

I checked that these registers are defined in the SKL configuration file so the hide/show functionality should be working.  Is it possible to attach a verbose log file to this issue so that I can debug why the functionality is not working?

elicn, 2019-02-25:
Sure; I pulled the latest changes and ran: `sudo ./chipsec_util.py --platform SKL mmio list --verbose`
Output attached: [mmio_list.log](https://github.com/chipsec/chipsec/files/2900772/mmio_list.log)

ErikBjorge, 2019-02-28:
The issue is that the SKL configuration is not compatible with the skylake server (SKX) processor (DID 0x2020).  The PCI configuration space data is going to be wrong and this is going to lead to problems when attempting to get base addresses.  We need to get the SKX configuration data released so your platform can be supported.

elicn, 2019-02-28:
For what it's worth, it didn't work on BDX either... but I haven't checked whether MMCFG is configured the same way on BDX and SKX.

ErikBjorge, 2019-02-28:
I will see if I can get hold of a system and have someone test it.  I will fix the PCH files as that may also be part of the issue on BDX.

Sae86, 2022-06-02:
Closing. Fix in #578. Feel free to reopen if needed.

closed 2022-06-02 by Sae86