chipsec/chipsec #529: Cnfusion on chipsec SMI implementation in driver code

bikashdash, 2019-02-10:
Hello Gentlemen. I asked this question in Google groups too.
i was doing some research on chipsec SMI driver and tried to understand the some of the concept. My main goal was how it is reaching to smi  ioctl call.
 I was actually trying to see smm_ptr module in chipsec  to understand how it reaches to the driver and make IOCTL call. so there is a win32helper.py who makes it happen to reach into driver with the help of IOCTL_SWSMI value defined in chipsec.h . so my question is that, is it really doing any IOCTL call?
Because i saw the c++ code involved in driver and i did not find any clue. let's see the below code which is actually involved in driver(driver.c line no 946)
case IOCTL_SWSMI:
          {
            CPU_REG_TYPE gprs[6] = {0};
            CPU_REG_TYPE _rax = 0, _rbx = 0, _rcx = 0, _rdx = 0, _rsi = 0, _rdi = 0;
            unsigned int _smi_code_data = 0;

            DbgPrint("[chipsec] > IOCTL_SWSMI\n");
			
            pInBuf = Irp->AssociatedIrp.SystemBuffer;
            if( !pInBuf )
              {
	        DbgPrint( "[chipsec] ERROR: NO data provided\n" );
                Status = STATUS_INVALID_PARAMETER;
                break;
              }
            if( IrpSp->Parameters.DeviceIoControl.InputBufferLength < sizeof(UINT16) + sizeof(gprs) )
              {
                DbgPrint( "[chipsec] ERROR: STATUS_INVALID_PARAMETER (input buffer size < sizeof(UINT16) + sizeof(gprs))\n" );
                Status = STATUS_INVALID_PARAMETER;
                break;

if you see the above code "[chipsec] > IOCTL_SWSMI\n"" have been written. And i can't see this in console.

below are some of output
>python chipsec_main.py -m tools.smm.smm_ptr -a fuzz 0x0:0x1

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6
[CHIPSEC] Arguments: -m tools.smm.smm_ptr -a fuzz 0x0:0x1

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API

[CHIPSEC] OS      : Windows 10 10.0.17763 AMD64
[CHIPSEC] Platform: Mobile 7th Generation Core Processor (Kabylake H)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5910
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A154

[+] loaded chipsec.modules.tools.smm.smm_ptr
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.smm.smm_ptr
[*] Module arguments (2):
['fuzz', '0x0:0x1']
[x][ =======================================================================
[x][ Module: A tool to test SMI handlers for pointer validation vulnerabilies
[x][ =======================================================================
Usage: chipsec_main -m tools.smm.smm_ptr [ -a <mode>,<config_file>|<smic_start:smic_end>,<size>,<address> ]
  mode          SMI handlers testing mode
    = config    use SMI configuration file <config_file>
    = fuzz      fuzz all SMI handlers with code in the range <smic_start:smic_end>
    = fuzzmore  fuzz mode + pass '2nd-order' pointers within buffer to SMI handlers
  size          size of the memory buffer (in Hex)
  address       physical address of memory buffer to pass in GP regs to SMI handlers (in Hex)
    = smram     pass address of SMRAM base (system may hang in this mode!)

[*] Allocated memory buffer (to pass to SMI handlers)       : 0x0000000075A9F000

[*] Allocated 2nd buffer (address will be in the 1st buffer): 0x0000000075A9E000

[*] Configuration
    SMI testing mode          : fuzz
    Range of SMI codes (B2)   : 0x00:0x01
    Memory buffer pointer     : 0x0000000075A9F000 (address passed in GP regs to SMI)
    Filling/checking memory?  : YES
      Second buffer pointer   : 0x0000000075A9E000 (address written to memory buffer)
      Number of bytes to fill : 0x500
      Byte to fill with       : 0x11
    Additional options (can be changed in the source code):
      Fuzzing SMI functions in ECX?          : 1
      Max value of SMI function in ECX       : 0x10
      Max value of SMI data (B3)             : 0x100
      Max offset of the pointer in the buffer: 0x20
      Passing pointer in all GP registers?   : 0
      Default values of the registers        : 0x5A5A5A5A5A5A5A5A
      Dump all register values every SMI     : 1
      Bail on first detection                : 1

[*] >>> Fuzzing SMI handlers..
[*] AX in RAX will be overwridden with values of SW SMI ports 0xB2/0xB3
    DX in RDX will be overwridden with value 0x00B2

[*] Setting values of general purpose registers to 0x5A5A5A5A5A5A5A5A

[*] reloading buffer with PTR at offset 0x0..
[*] writing 0x500 bytes at 0x0000000075A9F000

[*] fuzzing SMI# 0x00 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x0000000075A9F000
    > SMI 00 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x0000000075A9F000
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A

    < checking buffers

    RSI: 0x0000000075A9F000
    > SMI 00 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x5A5A5A5A5A5A5A5A
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x0000000075A9F000
      RDI: 0x5A5A5A5A5A5A5A5A

and going on..

if you see i can't see the print "IOCTL" in the console.
can anyone here verify that will be helpfull

hardboost, 2019-02-11:
You should use [DebugView](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview) to see those messages, because they are from kernel.

bikashdash, 2019-02-11:
I tried using Debugview which seems to not working . there is a issue with windows 10 using debugview. the error is "Could not extract Debug View driver to c:\Windows\System32\Drivers\Dbgv.sys..."
[image]

Sae86, 2022-05-09:
Hi @bikashdash, were you able to see the messaged in DebugView?

npmitche, 2024-03-12:
Hi @bikashdash, Let me try to clarify some of this. The IOCTL that is being sent is what allows the python code to communicate to the kernel driver. (win32helper.py > chipsec_hlpr.sys).  The C code in the kernel driver does not then initiate another IOCTL, it just has to decode which IOCTL was called.  That is why it's simply a case in the switch statement. 

For DebugView, be sure to run it as Administrator. 

I'm going to close this issue, feel free to reach out if you have other issues or questions.

closed 2024-03-12 by npmitche