chipsec/chipsec #531: Chipset Flash Region 6 (FREG6) bug in chipsec or chipset specification?

wrffrz, 2019-02-13:
According to the Series 100, 200, and 300 chipset datasheets available at:

https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/100-series-chipset-datasheet-vol-2.pdf
https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/200-series-chipset-pch-datasheet-vol-2.pdf
https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/300-series-chipset-pch-datasheet-vol-2.pdf

...the Flash Region 6 (FREG6) register does not exist.

However, the chipsec skl.xml, kbl.xml, pch_1xx.xml and pch_2xx.xml definition files include this register.

Have these registers been removed and it's a bug in chipsec or is there a bug in the chipset specifications?

ErikBjorge, 2019-02-13:
This is a bug in the configuration files.  We will submit a patch for this.

ErikBjorge, 2019-02-13:
Please check out PR #532 and PR #533 to see if they resolve this issue.

BrentHoltsclaw, 2019-10-10:
Issues appear to be fixed.  Please reopen if the issue persists.

closed 2019-10-10 by BrentHoltsclaw