chipsec/chipsec #535: Fails to build under kernel 5.0-rc6

superm1, 2019-02-20:
Receiving this error while trying to build.

```
building the linux driver
Makefile:189: ================= WARNING ================
Makefile:190: 'SUBDIRS' will be removed after Linux 5.3
Makefile:191: Please use 'M=' or 'KBUILD_EXTMOD' instead
Makefile:192: ==========================================
Makefile:189: ================= WARNING ================
Makefile:190: 'SUBDIRS' will be removed after Linux 5.3
Makefile:191: Please use 'M=' or 'KBUILD_EXTMOD' instead
Makefile:192: ==========================================
src/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec_km.c: In function ‘d_ioctl’:
src/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec_km.c:845:66: error: macro "access_ok" passed 3 arguments, but takes just 2
   if (!access_ok(VERIFY_READ, ioctl_param, sizeof(unsigned short))) {
                                                                  ^
src/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec_km.c:845:8: error: ‘access_ok’ undeclared (first use in this function)
   if (!access_ok(VERIFY_READ, ioctl_param, sizeof(unsigned short))) {
        ^~~~~~~~~
src/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec_km.c:845:8: note: each undeclared identifier is reported only once for each function it appears in
```

ErikBjorge, 2019-02-20:
What distro are you using?  Just want to make sure we are reproducing the failure in the same environment.  Looks like we have some other changes that need to be made as well to fix the SUBDIRS issue.

superm1, 2019-02-20:
Ubuntu 18.04.  I'm using 5.0-rc6 from the mainline PPA: https://kernel.ubuntu.com/~kernel-ppa/mainline/

BrentHoltsclaw, 2019-03-20:
Issue should be resolved with PR #536.  Was merged and is within Chipsec 1.3.7.  If other problems occur please open a new issue. Thanks.

closed 2019-03-20 by BrentHoltsclaw