chipsec/chipsec #543: Chipsec doesn't correctly parse Lenovo Yoga 2 Pro UEFI firmware image

wrffrz, 2019-03-22:
I've got a Lenovo Yoga 2 Pro which is a Hawell laptop and chipsec has a couple problems decoding the firmware image from this system.

First, when we run chipsec_util.py to decode the image, it crashes trying to parse a variable name using utf16:

> $ python /nas/jesse/tools/chipsec/chipsec_util.py decode spidump.bin
>
> ################################################################
> ##                                                            ##
> ##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
> ##                                                            ##
> ################################################################
> [CHIPSEC] Version 1.3.7
> [CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
> [CHIPSEC] Executing command 'decode' with args ['spidump.bin']
>
> [CHIPSEC] Decoding SPI ROM image from a file 'spidump.bin'
> [CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary 'spidump.bin'
> Traceback (most recent call last):
>   File "/nas/jesse/tools/chipsec/chipsec_util.py", line 232, in <module>
>     sys.exit( main() )
>   File "/nas/jesse/tools/chipsec/chipsec_util.py", line 228, in main
>     return chipsecUtil.main()
>   File "/nas/jesse/tools/chipsec/chipsec_util.py", line 195, in main
>     comm.run()
>   File "/nas/jesse/tools/chipsec/chipsec/utilcmd/decode_cmd.py", line 122, in run
>     spi_uefi.decode_uefi_region(_uefi, pth, fname, fwtype)
>   File "/nas/jesse/tools/chipsec/chipsec/hal/spi_uefi.py", line 558, in decode_uefi_region
>     _uefi.parse_EFI_variables( nvram_fname, region_data, False, fwtype )
>   File "/nas/jesse/tools/chipsec/chipsec/hal/uefi.py", line 358, in parse_EFI_variables
>     decode_EFI_variables( efi_vars, nvram_pth )
>   File "/nas/jesse/tools/chipsec/chipsec/hal/uefi.py", line 234, in decode_EFI_variables
>     parse_efivar_file(var_fname, data, AUTH_SIG_VAR)
>   File "/nas/jesse/tools/chipsec/chipsec/hal/uefi_common.py", line 1021, in parse_efivar_file
>     parse_auth_var(var, var_path)
>   File "/nas/jesse/tools/chipsec/chipsec/hal/uefi_common.py", line 971, in parse_auth_var
>     var_name = codecs.decode(db[tof:tof+name_size], 'utf-16')
>   File "/usr/lib/python2.7/encodings/utf_16.py", line 16, in decode
>     return codecs.utf_16_decode(input, errors, True)
> UnicodeDecodeError: 'utf16' codec can't decode bytes in position 28-29: illegal encoding
> $

Although we get this error from chipsec, we're still able to extract 43 .TE files and 14 .PE32 files from the image:

> $ find spidump.bin.dir -iname '*.te' -o -iname '*.pe32' -o -iname '*.efi' | wc -l
> 58
> $

However, it actually has a lot more than that.

One of the sections that chipsec identifies as a RAW section is 12MB and contains a UEFI Firmware Volume:

> $ find . -name '*_S_RAW' -ls
>  73382072      1 -rw-rw-r--   1 jesse    jesse          76 Mar 22 22:31 ./1_400000-7FFFFF_BIOS.bin.dir/FV/02_7A9354D9-0468-444A-81CE-0BF617D890DF.dir/60_1BA0062E-C779-4582-8566-336AE8F78F09.FV_SECURITY_CORE.dir/02_S_RAW
>  73382067      1 -rw-rw-r--   1 jesse    jesse           0 Mar 22 22:31 ./1_400000-7FFFFF_BIOS.bin.dir/FV/02_7A9354D9-0468-444A-81CE-0BF617D890DF.dir/60_1BA0062E-C779-4582-8566-336AE8F78F09.FV_SECURITY_CORE.dir/00_S_RAW
>  73381309   3549 -rw-rw-r--   1 jesse    jesse    12582924 Mar 22 22:31 ./1_400000-7FFFFF_BIOS.bin.dir/FV/00_7A9354D9-0468-444A-81CE-0BF617D890DF.dir/00_4A538818-5AE0-4EB2-B2EB-488B23657022.FV_DXE_CORE.dir/00_S_COMPRESSION.dir/00_S_RAW
> $ xxd ./1_400000-7FFFFF_BIOS.bin.dir/FV/00_7A9354D9-0468-444A-81CE-0BF617D890DF.dir/00_4A538818-5AE0-4EB2-B2EB-488B23657022.FV_DXE_CORE.dir/00_S_COMPRESSION.dir/00_S_RAW | head -10
> 00000000: 3132 3334 3536 3738 3941 0d0a 0000 0000  123456789A......
> 00000010: 0000 0000 0000 0000 0000 0000 d954 937a  .............T.z
> 00000020: 6804 4a44 81ce 0bf6 17d8 90df 0000 c000  h.JD............
> 00000030: 0000 0000 5f46 5648 ff8e ffff 4800 b14b  ...._FVH....H..K
> 00000040: 0000 0001 0300 0000 0000 4000 0000 0000  ..........@.....
> 00000050: 0000 0000 504f b7ab 2dfd 7240 a321 cafc  ....PO..-.r@.!..
> 00000060: 7297 7efa a284 0640 220e 00f8 0600 001b  r.~....@".......
> 00000070: 0708 0000 e40d 0010 4d5a 0000 0000 0000  ........MZ......
> 00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................
> $

If we then use chipsec_util.py uefi decode to extract that 00_S_RAW section, we get 234 more EFI executables:

> $ cp ./1_400000-7FFFFF_BIOS.bin.dir/FV/00_7A9354D9-0468-444A-81CE-0BF617D890DF.dir/00_4A538818-5AE0-4EB2-B2EB-488B23657022.FV_DXE_CORE.dir/00_S_COMPRESSION.dir/00_S_RAW .
> $ python /nas/jesse/tools/chipsec/chipsec_util.py -n uefi decode 00_S_RAW
>   ...
> $ find 00_S_RAW.dir -iname '*.efi' | wc -l 
>     234
> $

wrffrz, 2019-03-22:
[spidump.zip](https://github.com/chipsec/chipsec/files/2998730/spidump.zip)

ErikBjorge, 2019-03-23:
Thanks for the report and providing the failing flash image.  We will take looks at this and see what it will take to fix.  Also appreciate the debug information in the issue report.

BrentHoltsclaw, 2019-03-25:
Looked into why the image is not being extracted properly and found that the section type is unknown/reserved, which means CHIPSEC is not doing anything to try to decode it.  Can add logic to try and decode unknown/reserved sections, but should that be default behavior?

For the decode issue will put in a try, except and print out a warning which will be within the uefi.lst file.

hughsie, 2019-03-27:
If it helps, master also fails for firmware.bin inside b76ea728bd1e305c285fbda742159c92a8d44924-Lenovo-ThinkPad-T480s-SystemFirmware-1.30.cab -- this time with `SPI Flash descriptor region is not valid` -- if you want a *ton* of uefi firmware images for testing please just ask and I can send you links.

BrentHoltsclaw, 2019-03-27:
A collection of uefi firmware images would be helpful, especially if you have some known issues.

hughsie, 2019-03-27:
Okay, here's a long list:

    145f75656a32ac86daecab9c74730f10c74895c7-Lenovo-ThinkPad-T580P52s-SystemFirmware-0.1.14.cab
    4a08092190ffa09fed1a7375af9402273b52d804-Lenovo-ThinkPad-A475-SystemFirmware-1.14.cab
    d537d51a050c61f7fbb8d38ac266526dbd133166-Lenovo-ThinkPad-A275-SystemFirmware-1.15.cab
    70fb868eee2d1a6480128c908c880d758a6ff3f1-Lenovo-ThinkPad-P1X1Extreme-SystemFirmware-0.1.15.cab
    ab85ae6b34b6ef1fd9d556e56ec6c48fa8c69909-Lenovo-ThinkPad-T480-SystemFirmware-1.16.cab
    28f665a9bae6024e286c1dc9d016e3a4ad0dbdd3-Lenovo-ThinkPad-P52P72-SystemFirmware-1.17.cab
    36e2b22d7296403e8d0ca51b74058af18048d8af-Lenovo-ThinkPad-X280-SystemFirmware-1.19.cab
    f56458ba52f3c0bca4ab883e9b8a112b970ef5f7-Lenovo-ThinkPad-P71-SystemFirmware-1.24.cab
    dfc6d62e3e26d4d54a656a36c00768985179ce7d-Lenovo-ThinkPad-T470p-SystemFirmware-1.24.cab
    064c5aa3256d43f5ed7c931447c20418203fab2f-Lenovo-ThinkPad-X1Yoga3nd-SystemFirmware-0.1.24.cab
    2745a26ac883b9c171522bb16e1c628632932b20-Lenovo-ThinkPad-T480s-SystemFirmware-1.25.cab
    769be80680bf15ee51a3782e664041900c45de76-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.27.cab
    c910d903f6d7ac628ce5bdd5285824dee60ef397-Lenovo-ThinkPad-T470s-SystemFirmware-1.28.cab
    eef567d2363ca313d077c679ac379e52c2334fd4-Lenovo-ThinkPad-X1Yoga2nd-SystemFirmware-0.1.29.cab
    e867f209ab9af8a055a53d578c82228368a36144-Lenovo-ThinkPad-T560P50s-SystemFirmware-1.29.cab
    292eed28994234bf591e24addd4402d4135525e8-Lenovo-ThinkPad-L480-SystemFirmware-1.29.cab
    931b215c56ee380e4bd7dd611baccc46851e2cfb-Lenovo-ThinkPad-X270-SystemFirmware-1.32.cab
    2b2f0a343bdc7a2861e20779c3f12f9bd3c46d5f-Lenovo-ThinkPad-X1Carbon5th-SystemFirmware-1.34.cab
    dd1fdf1ac801f502ca5e00a3a79c025544704474-Lenovo-ThinkPad-T570P51s-SystemFirmware-0.1.35.cab
    6c4063e5972ee0de9870f1b63b7fd7310a0c179d-Lenovo-ThinkPad-L570-SystemFirmware-0.1.35.cab
    46c382fbcc171b023b66b0f579a7f7ff31d90ff9-Lenovo-ThinkPad-T460s-SystemFirmware-1.37.cab
    8366a0a2f694f2e67e8d629c3cd5b078c10d6535-Lenovo-ThinkPad-T460-SystemFirmware-1.37.cab
    3c6d0c91e496c35435e6165b660ed2d3d3510415-Lenovo-ThinkPad-X1Carbon4th-X1Yoga1st-SystemFirmware-1.38.cab
    f9abd647920da7d0b9d178ed3bfb93516cc9e390-Lenovo-ThinkPad-X260-SystemFirmware-1.40.cab
    9833ac83b1454a6391613d6927ecdc38e53e7a94-Lenovo-ThinkPad-T460s-SystemFirmware-1.42.cab
    9d9750f0e6d5c378cdd8f7eecde0eb8538a98c19-Lenovo-ThinkPad-L560-SystemFirmware-1.43.cab
    ba8248caf4f6a9a00080c75eabf09c7bf429abc0-Lenovo-ThinkPad-P51-SystemFirmware-1.45.cab
    886b60736445329b5ddc1797ddb013c0b49501f4-Lenovo-ThinkPad-T470-SystemFirmware-1.52.cab
    1c03e1b5491094429129b3eef952816c21777281-Lenovo-ThinkPad-P50-SystemFirmware-1.52.cab
    f91876dff5d0ac439364e466814cee8c5df01a4a-Lenovo-ThinkPad-L470-SystemFirmware-1.68.cab
    995cc47a2f491d6fc700089984da6662bcca17da-Lenovo-ThinkPad-P70-SystemFirmware-2.25.cab
    f617a912dc6eba4766de675bffe1e58a7ca9bae9-Lenovo-ThinkPad-T460p-SystemFirmware-2.26.cab
    071423c257f8305d67cd81fddbbc644046988fbd-M60_0241.cab
    3c9bf9b60b418e9914c9e8a5a63853ebf6fbb6d6-firmware.XPS_7760_Precision_5720_AIO_2.2.0.wu.cab
    d53c52f3a62ef1230ffd306dbc20a375cc9d2141-firmware_Precision_Tower_3620_2.3.5.wu.cab
    f4f334a82d19f40cc7ce7416f15e68702872184c-firmware_Precision_Tower_3420_2.3.5.wu.cab
    a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab
    6573e95f3482252d28863d1ed199c44aa0fa377f-Lenovo-ThinkPad-T580andP52s-ConsumerMEFirmware-11.8.55.3510.cab
    97e72530051a8c17626e7168e5a007cfac07b6a7-Lenovo-ThinkPad-T580andP52s-CorporateMEFirmware-11.8.55.3510.cab
    9748e46730f66d51aae658e3eb24acc4e3ac9ab1-Lenovo-ThinkPad-X1Yoga3rd-ConsumerMEFirmware-11.8.55.3510.cab
    6089fae9264d8649b572e118fc66350e16464945-Lenovo-ThinkPad-X1Yoga3rd-CorporateMEFirmware-11.8.55.3510.cab
    530d6cde82f5152396edcff4099fefe3e98072a9-Lenovo-ThinkPad-T570andP51s-ConsumerMEFirmware-11.8.55.3510.cab
    e31314a77b858ac4478a29e1665df2e4d539ec36-Lenovo-ThinkPad-T570andP51s-CorporateMEFirmware-11.8.55.3510.cab
    3703fa845dcb19e5f7ae77bdab70479970c2569d-Lenovo-ThinkPad-T480s-ConsumerMEFirmware-11.8.55.3510.cab
    3da421eb27b392a3941c09b1cf7976ccd3c28b1e-Lenovo-ThinkPad-T480s-CorporateMEFirmware-11.8.55.3510.cab
    9c0d1765507c95851b31d5b4b3f139eb22115f1d-Lenovo-ThinkPad-X1Carbon6th-ConsumerMEFirmware-11.8.55.3510.cab
    dd546ac2dc869a1b7dc66d0eefe1636124e0727c-Lenovo-ThinkPad-X1Carbon6th-CorporateMEFirmware-11.8.55.3510.cab
    d854a371fccaa0304e001e10598ee82d88e0232f-Lenovo-ThinkPad-T480-ConsumerMEFirmware-11.8.55.3510.cab
    a2e10f4044223318f66d18015f9e12ae23b6faf9-Lenovo-ThinkPad-T480-CorporateMEFirmware-11.8.55.3510.cab
    c46722e51001648be9ba0a451c6e1498f154d0ff-Lenovo-ThinkPad-P51-ConsumerMEFirmware-11.8.55.3510.cab
    b2f92767d757abd51cee851f097cb3cff5ebc220-Lenovo-ThinkPad-P51-CorporateMEFirmware-11.8.55.3510.cab
    bd160ccdaa40a6db2b310045d6652f7e60c2bf5d-Lenovo-ThinkPad-P71-ConsumerMEFirmware-11.8.55.3510.cab
    d6a0d17380c25470b515f9f18ccdb08b10ec1bde-Lenovo-ThinkPad-P71-CorporateMEFirmware-11.8.55.3510.cab
    fbf02e9ec3b8b5c652083c2ebe5d1d04537998f2-Lenovo-ThinkPad-T470-ConsumerMEFirmware-11.8.55.3510.cab
    a8bf4cd55b3e10898e6d7dc48bad96bc5247e6b8-Lenovo-ThinkPad-T470-CorporateMEFirmware-11.8.55.3510.cab
    87259b19bf31731007132ed99178876b439a772c-Lenovo-ThinkPad-T470s-ConsumerMEFirmware-11.8.55.3510.cab
    7ec8fb79882ca0d589b3c0d5552b5dd36599b667-Lenovo-ThinkPad-T470s-CorporateMEFirmware-11.8.55.3510.cab
[... cut ...]
    6215e95bd8c53dddad1742d54e0a0299b6aeb54f-firmware_Precision_7x10_1.12.4.wu.cab
    f7375df3c5f903f55ffd64e9ce891da3aa535355-DellTpm1.2_Fw5.81.2.1.cab

Each can be prepended with `https://fwupd.org/downloads/` if you want to download it. You probably only need one ThinkPad firmware, for example, but I don't mind if you want them all to do some kind of validation. The cab files can be extracted using `gcab -x` but you'll need to create a directory for each archive as inside most of the firmware capsules will be called `firmware.bin`. If you need a hand, ask.

ErikBjorge, 2019-03-27:
Wow, thanks for the list and providing the link.  We will be looking at them.

BrentHoltsclaw, 2019-04-02:
@hughsie can you please review pull request #544, to verify that it handles your concerns.

hughsie, 2019-04-02:
@BrentHoltsclaw using your branch I still get:

[CHIPSEC] Found SPI Flash descriptor at offset 0x858891 in the binary 'firmware.bin'
ERROR: SPI Flash descriptor region is not valid

That's using firmware.bin from b76ea728bd1e305c285fbda742159c92a8d44924-Lenovo-ThinkPad-T480s-SystemFirmware-1.30.cab

BrentHoltsclaw, 2019-04-02:
Which util command are you using?  I've been working on fixing chipsec_util.py uefi decode binaryname.  It appears that chipsec_util.py decode binaryname is indeed broken at this time.  I'll look into fixing that one up as well.

hughsie, 2019-04-02:
`chipsec_util decode firmware.bin -n`

BrentHoltsclaw, 2019-04-02:
After a discussion with @ErikBjorge chipsec_util decode expects to find a valid spi flash descriptor.  I updated the logging to indicate that chipsec_util uefi decode should be used when the spi flash descriptor is invalid or cannot be found.  Sound reasonable?

hughsie, 2019-04-03:
> chipsec_util uefi decode

Works for me! Now all we need is for Python 3 support to be merged and I can do something interesting on the LVFS :)

frinzell, 2021-02-22:
Closing as fixed.

closed 2021-02-22 by frinzell