chipsec/chipsec #549: Kabylake DID 0x5918 added twice

gibarrasch, 2019-04-11:
Hi, I see the DID '0x5918' twice, 
 - First with the description: "Mobile 7th Generation Core Processor (Kabylake H)" (https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L177) 
 
 - And then with the description "Intel Xeon Processor E3 v6 (Kabylake CPU)" (https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L209)

ErikBjorge, 2019-04-11:
Nice catch.  This should not cause any functional issues because the both point to the KBL platform configuration.  It does need to be cleaned up.  I will check to what description matches the actual ID (most likely it is the Xeon description).

frinzell, 2021-02-22:
Closing as fixed.  Please reopen if needed.

closed 2021-02-22 by frinzell