chipsec/chipsec #550: MMCFG in VTBAR

gibarrasch, 2019-04-16:
Hi, Sorry this is not an actual issue I just want to understand better the need of MMCFG type of register, i saw for instance that the VTBAR definition is override for old servers as follow:

 bdx.xml
<register name="VTBAR" type="mmcfg" bus="0" dev="5" fun="0" offset="0x180" size="4" desc="VT BAR Register">

hsx.xml
<register name="VTBAR" type="mmcfg" bus="0" dev="5" fun="0" offset="0x180" size="4" desc="VT BAR Register">

ivt.xml
<register name="VTBAR" type="pcicfg" bus="0" dev="5" fun="0" offset="0x180" size="4" desc="VT BAR Register">

jkt.xml
<register name="VTBAR" type="pcicfg" bus="0" dev="5" fun="0" offset="0x180" size="4" desc="VT BAR Register">

Why bdx and hsx need to use the memory mapped instead of doing a direct pcicfg access, whereas jkt and ivt can perform a direct pci read? does it have a reason or both options can be used interchangeable?

gibarrasch, 2019-04-16:
Is it because pcicfg only allow us to read the PCI configuration space (the first 256 bytes)? if it is, i assume that ivt and jkt should be mmcfg instead. am i right?

frinzell, 2019-04-19:
Good catch.  IVT and JKT XML should be 'mmcfg'.

frinzell, 2019-04-19:
PR #551 created and ready for review.

BrentHoltsclaw, 2019-10-10:
This issue appears to be fixed.  Please reopen if farther clarification is needed or the issue persists.

closed 2019-10-10 by BrentHoltsclaw