chipsec/chipsec #553: MMCFG miss-definitions

gibarrasch, 2019-05-06:
Hi Guys, about MMCFG registers I think there are some miss-definitions for some architectures:

**Avoton:**
It doesn't define any override so it uses the common.xml definition of MMCFG bar:

common.xml

_bar name="MMCFG"    bus="0" dev="0"    fun="0" reg="0x60" width="8" mask="0x7FFFFFF000" size="0x1000" enable_bit="0" desc="PCI Express Register Range"_

But I think it should define and use the BECREG as the other atoms:

_bar name="MMCFG"   register="BECREG" base_field="ECBASE" size="0x10000000" enable_bit="ECENABLE" desc="PCI Express Register Range"_

**Ivytown and Jaketown:**
It doesn't define any override so it uses the common.xml definition of MMCFG bar too.

Looking into the Xeon E5 v2 datasheet.

mmcfg_base is in bus: 0, device: 5, function: 0, offset 0x84.

But, I couldn't find datasheets to verify it for every Ivytown and Jaketown.

ErikBjorge, 2019-05-08:
We will have to dig into the documentation to verify the correct registers.  Thanks for finding these issues.

Sae86, 2024-04-16:
Haven't found any documents for Avoton but I agree it should be consistent among Atoms.

PR #2132 (2024-04-22): Update MMCFG for Avoton to be consistent with other Atoms
Fixes #553
closed 2024-04-22 by npmitche