chipsec/chipsec #560: ./setup.py build does not build LZMA decompressor

hughsie, 2019-05-14:
on master, in a fresh checkout:

    $ ./setup.py build
    running build
    running build_py
    wget https://fwupd.org/downloads/4689401be87da70d7ea5685c0941e42a4df40830-Lenovo-ThinkPad-P50-SystemFirmware-1.56.cab
    gcab -x 4689401be87da70d7ea5685c0941e42a4df40830-Lenovo-ThinkPad-P50-SystemFirmware-1.56.cab
    ./chipsec_util.py -n uefi decode firmware.bin
    ....
      File "/home/hughsie/Code/chipsec/chipsec/helper/linux/helper.py", line 1133, in decompress_file
        data = subprocess.check_output(decode_str,shell=True)
      File "/usr/lib64/python2.7/subprocess.py", line 223, in check_output
        raise CalledProcessError(retcode, cmd, output=output)
    subprocess.CalledProcessError: Command 'chipsec_tools/compression/bin/LzmaCompress -d -o sect00_rxxt sect00_rxxt.gz' returned non-zero exit status 127

Sure enough:

    $ ls chipsec_tools/compression/bin/LzmaCompress
    ls: cannot access 'chipsec_tools/compression/bin/LzmaCompress': No such file or directory

hughsie, 2019-05-14:
Of course, a workaround is `cd chipsec_tools/compression/ && make` but I'm guessing that's not what was expected.

BrentHoltsclaw, 2019-05-14:
I noticed the same issue yesterday.  It was supposed to be built within setup.py

BrentHoltsclaw, 2019-10-10:
Issue appears to be fixed.  Please reopen if this is not the case.

closed 2019-10-10 by BrentHoltsclaw